nefarius/ViGEmBus #3: WiFi HID emulator

mkeyno, 2018-08-19:
hi , sorry to open a ticket for my question, but I have a question that may related to your field of repository and any possible hint really appreciated
Actually I interested to send game-pad data by WiFi bridge(ESP8266) to my laptop, but generic joystick only use USB port, Is there a way to emulate "HID-compliant game controller" driver which listen to IP message instead of USB port
Tnx

mkeyno, 2018-08-19:
thanks for quick reply , I've actually prepared my server  side with ESP8266 which can send any joystick data (buttons,pot,..) on websocket message , so on my laptop wifi emulator should connect to specific IP and listen to websocket message,  and upon any mesege pass it to generic HID-compliant game controller
I'm not sure, how to bind wifi message with HID driver on laptop but I find [this site ](http://maxjoypad.com/)made similar software ,but unfortunately it is not opensource to modify it 
I hope I make it clear 
tnx

nefarius, 2018-08-20:
Well, when you already have an established method to transfer the data to your PC then all you need is to feed that data into a Gamepad emulator like ViGEm or vJoy.

nefarius, 2018-08-21:
I haven't yet found the time to update the [feeder documentation](https://docs.vigem.org/#!vigem-feeder.md) but explore the [sample implementations](https://github.com/nefarius/ViGEm/tree/master/Src/Samples) to get an idea on how to use the client library. Also a lot of action happens on the Discord so consider joining.

mkeyno, 2018-08-22:
@nefarius can you direct me where should I start and entry point for the ViGEm, my skill is mostly focus on Soc programming with c++ but I have  little skill on visual c#
tnx

nefarius, 2018-08-22:
The examples I linked to are in C++...

closed 2018-09-22 by nefarius