nefarius/ViGEmBus #5: Hot to emulate ps4 controller to play ps4 remote play?

Frequel, 2018-09-24:
sorry for my question but i didn't find the discord server on the read.me .
i have xbox one gamepad and DS4 gamepad but i don't have the DS4 usb wireless adaptor, so i was find a solution to use my DS4 without the adapter but with my bluetooth dongle.
i followed the installation instruction of vigembus but i don't how to reach one of the purpouse of this git :
"You want the freedom to use a different controller of your choice in PS4 Remote Play."
can you help me please?

nefarius, 2018-09-24:
The links to both Discord and Forums are all [on the main page](https://vigem.org/) and the chat badge in the README.

I'll close this here since it's neither a bug report nor a feature request.

closed 2018-09-24 by nefarius

Frequel, 2018-09-24:
ok sorry, in the readme i couldn;t find anything, in the website i find the discord link right now. thank you