nefarius/ViGEmBus #6: UNEXPECTED_KERNEL_MODE_TRAP Bluescreen

JamieH, 2018-09-29:
**Describe the bug**
While playing Forza Horizon 4 my machine bluescreened

**To Reproduce**
Not sure on the exact way to reproduce, Yesterday it worked fine *all day* and then today within an hour of playing I got hit by this bluescreen.

Here is the kernel minidump:
[092918-10375-01.zip](https://github.com/ViGEm/ViGEmBus/files/2430711/092918-10375-01.zip)

**Expected behavior**
No bluescreen

**Screenshots**
NA

**System details (please complete the following information):**
 - OS: Windows 10 Pro 10.0.17134
 - Feeder software: https://github.com/Alia5/GloSC 2.0.5 cc @Alia5
 - Driver Version: 1.14.3.0

**Additional context**
```
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

UNEXPECTED_KERNEL_MODE_TRAP (7f)
This means a trap occurred in kernel mode, and it's a trap of a kind
that the kernel isn't allowed to have/catch (bound trap) or that
is always instant death (double fault).  The first number in the
bugcheck params is the number of the trap (8 = double fault, etc)
Consult an Intel x86 family manual to learn more about what these
traps are. Here is a *portion* of those codes:
If kv shows a taskGate
        use .tss on the part before the colon, then kv.
Else if kv shows a trapframe
        use .trap on that value
Else
        .trap on the appropriate frame will show where the trap was taken
        (on x86, this will be the ebp that goes with the procedure KiTrap)
Endif
kb will then show the corrected stack.
Arguments:
Arg1: 0000000000000008, EXCEPTION_DOUBLE_FAULT
Arg2: ffffab817b00a150
Arg3: ffff86860dfd0f60
Arg4: fffff802c02442df

Debugging Details:
------------------

DUMP_CLASS: 1

DUMP_QUALIFIER: 400

BUILD_VERSION_STRING:  17134.1.amd64fre.rs4_release.180410-1804

SYSTEM_MANUFACTURER:  System manufacturer

SYSTEM_PRODUCT_NAME:  System Product Name

SYSTEM_SKU:  SKU

SYSTEM_VERSION:  System Version

BIOS_VENDOR:  American Megatrends Inc.

BIOS_VERSION:  0403

BIOS_DATE:  09/26/2017

BASEBOARD_MANUFACTURER:  ASUSTeK COMPUTER INC.

BASEBOARD_PRODUCT:  ROG STRIX Z370-H GAMING

BASEBOARD_VERSION:  Rev 1.xx

DUMP_TYPE:  2

BUGCHECK_P1: 8

BUGCHECK_P2: ffffab817b00a150

BUGCHECK_P3: ffff86860dfd0f60

BUGCHECK_P4: fffff802c02442df

BUGCHECK_STR:  0x7f_8

TRAP_FRAME:  ffffab817b00a150 -- (.trap 0xffffab817b00a150)
NOTE: The trap frame does not contain all registers.
Some register values may be zeroed or incorrect.
rax=ffffd004aaff0ccb rbx=0000000000000000 rcx=ffffab817b000180
rdx=ffff8789fb608080 rsi=0000000000000000 rdi=0000000000000000
rip=fffff802c02442df rsp=ffff86860dfd0f60 rbp=ffff86860dfd1031
 r8=ffff86860dfd10d0  r9=ffff86860dfd10c4 r10=ffffab817b000180
r11=0000000000000000 r12=0000000000000000 r13=0000000000000000
r14=0000000000000000 r15=0000000000000000
iopl=0         nv up ei ng nz na pe nc
nt!KiChooseTargetProcessor+0x2f:
fffff802`c02442df 4c894da7        mov     qword ptr [rbp-59h],r9 ss:0018:ffff8686`0dfd0fd8=????????????????
Resetting default scope

CPU_COUNT: c

CPU_MHZ: e70

CPU_VENDOR:  GenuineIntel

CPU_FAMILY: 6

CPU_MODEL: 9e

CPU_STEPPING: a

CPU_MICROCODE: 6,9e,a,0 (F,M,S,R)  SIG: 70'00000000 (cache) 70'00000000 (init)

CUSTOMER_CRASH_COUNT:  1

DEFAULT_BUCKET_ID:  WIN8_DRIVER_FAULT

PROCESS_NAME:  ForzaHorizon4.

CURRENT_IRQL:  2

ANALYSIS_SESSION_HOST:  DESKTOP-VBK1LOK

ANALYSIS_SESSION_TIME:  09-29-2018 15:01:22.0118

ANALYSIS_VERSION: 10.0.10586.567 x86fre

STACK_OVERFLOW: Stack Limit: ffff86860dfd1000. Use (kF) and (!stackusage) to investigate stack usage.

LAST_CONTROL_TRANSFER:  from fffff802c03bef69 to fffff802c03ae380

STACK_TEXT:  
ffffab81`7b00a008 fffff802`c03bef69 : 00000000`0000007f 00000000`00000008 ffffab81`7b00a150 ffff8686`0dfd0f60 : nt!KeBugCheckEx
ffffab81`7b00a010 fffff802`c03ba87f : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nt!KiBugCheckDispatch+0x69
ffffab81`7b00a150 fffff802`c02442df : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nt!KiDoubleFaultAbort+0x27f
ffff8686`0dfd0f60 fffff802`c0243670 : 006408d2`01006d00 1a49d201`00c44273 9a650201`40003e10 f0004e06`49171124 : nt!KiChooseTargetProcessor+0x2f
ffff8686`0dfd1090 fffff802`c0243313 : ffffab81`7b000180 fd220449`410c077d 003c3106`4941088c 7dfe077d`00000000 : nt!KiDeferredReadyThread+0x240
ffff8686`0dfd1150 fffff802`c024311f : ffff8789`fb6081f0 00000000`00000000 00f13e00`f1fe5500 63206f66`6e690d18 : nt!KiReadyThread+0x33
ffff8686`0dfd1180 fffff802`c0242111 : ed0200f5`212ef5fc 65636c65`00000000 ffffab81`7b000180 07526e60`e1020240 : nt!KiExitDispatcher+0xff
ffff8686`0dfd11f0 fffff802`c0241a97 : ffff8789`e3ce4e10 00000000`00000200 fffff802`a122de40 00000000`0000000a : nt!IopfCompleteRequest+0x661
ffff8686`0dfd1310 fffff802`a1189627 : ffff8789`ddac4740 ffff8789`e79c3e10 00000000`00000000 ffff8789`d93bec20 : nt!IofCompleteRequest+0x17
ffff8686`0dfd1340 fffff802`a11b2cfb : 00000000`00000001 00000000`00000000 00007876`00186628 00000000`ec393ac8 : Wdf01000!FxRequest::CompleteInternal+0x247 [minkernel\wdf\framework\shared\core\fxrequest.cpp @ 869]
ffff8686`0dfd1400 fffff802`a5772c49 : 00000000`0000000c ffff8789`ddac4740 ffff8686`0dfd14c0 00000000`0000000c : Wdf01000!imp_WdfRequestCompleteWithInformation+0x9b [minkernel\wdf\framework\shared\core\fxrequestapi.cpp @ 571]
ffff8686`0dfd1470 00000000`0000000c : ffff8789`ddac4740 ffff8686`0dfd14c0 00000000`0000000c 00000000`00000000 : ViGEmBus+0x2c49
ffff8686`0dfd1478 ffff8789`ddac4740 : ffff8686`0dfd14c0 00000000`0000000c 00000000`00000000 7865646e`49657461 : 0xc
ffff8686`0dfd1480 ffff8686`0dfd14c0 : 00000000`0000000c 00000000`00000000 7865646e`49657461 02490249`00490261 : 0xffff8789`ddac4740
ffff8686`0dfd1488 00000000`0000000c : 00000000`00000000 7865646e`49657461 02490249`00490261 75335859`6f312c22 : 0xffff8686`0dfd14c0
ffff8686`0dfd1490 00000000`00000000 : 7865646e`49657461 02490249`00490261 75335859`6f312c22 535f5849`784b364e : 0xc

STACK_COMMAND:  kb

THREAD_SHA1_HASH_MOD_FUNC:  85625f0957ec52a96cb59fdde7de0916a8ec6c01

THREAD_SHA1_HASH_MOD_FUNC_OFFSET:  ca0370ea1c8bdef62e5e9ba0b2b19af343425704

THREAD_SHA1_HASH_MOD:  b802136d69383a2aa5f21aef7abf1e3e0ef84681

FOLLOWUP_IP: 
ViGEmBus+2c49
fffff802`a5772c49 33c0            xor     eax,eax

FAULT_INSTR_CODE:  8b48c033

SYMBOL_STACK_INDEX:  b

SYMBOL_NAME:  ViGEmBus+2c49

FOLLOWUP_NAME:  MachineOwner

MODULE_NAME: ViGEmBus

IMAGE_NAME:  ViGEmBus.sys

DEBUG_FLR_IMAGE_TIMESTAMP:  5a624d2c

BUCKET_ID_FUNC_OFFSET:  2c49

FAILURE_BUCKET_ID:  0x7f_8_ViGEmBus!Unknown_Function

BUCKET_ID:  0x7f_8_ViGEmBus!Unknown_Function

PRIMARY_PROBLEM_CLASS:  0x7f_8_ViGEmBus!Unknown_Function

TARGET_TIME:  2018-09-29T13:44:55.000Z

OSBUILD:  17134

OSSERVICEPACK:  0

SERVICEPACK_NUMBER: 0

OS_REVISION: 0

SUITE_MASK:  272

PRODUCT_TYPE:  1

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

OSEDITION:  Windows 10 WinNt TerminalServer SingleUserTS

OS_LOCALE:  

USER_LCID:  0

OSBUILD_TIMESTAMP:  2018-09-15 03:18:09

BUILDDATESTAMP_STR:  180410-1804

BUILDLAB_STR:  rs4_release

BUILDOSVER_STR:  10.0.17134.1.amd64fre.rs4_release.180410-1804

ANALYSIS_SESSION_ELAPSED_TIME: 474

ANALYSIS_SOURCE:  KM

FAILURE_ID_HASH_STRING:  km:0x7f_8_vigembus!unknown_function

FAILURE_ID_HASH:  {696f4673-2ffc-9435-98a6-63b09e22f46d}

Followup:     MachineOwner
---------
```

Squall-Leonhart, 2018-09-29:
this version hasn't changed for months and it works on both my pc's, i suggest checking other device drivers on the systemi as well as the hardware itself.

ps: your bios is 12 months out of date.

nefarius, 2018-09-30:
Thanks for the detailed report! Are you by any chance running Software from Oculus?

nefarius, 2018-09-30:
@JamieH did this happen somewhere mid-game or close to an event like plugging in a controller?

JamieH, 2018-09-30:
Hey, this happened mid-game. I don't think anything happened like that.

nefarius, 2018-09-30:
Good to know, might have already found one or more cause(s).

nefarius, 2019-04-18:
[Fixed in v1.16.112](https://github.com/ViGEm/ViGEmBus/releases/tag/v1.16.112)

closed 2019-04-18 by nefarius