nefarius/ViGEmBus #10: vigem_target_add with x360 target hangs Windows 7

kozec, 2018-11-26:
**Describe the bug**
Calling `vigem_target_add` with x360 target hangs (freezes) Windows 7. There is no BSOD nor error message, system simply stops.

**To Reproduce**
Consider following sample code:

```
#include "utils/assert.h"
#include "windows.h"
#include "ViGEm/Client.h"
// g++ -Iinclude -Ivigemclient/include/ ViGEmClient.cpp c-sample.c -lsetupapi -lhid

static PVIGEM_CLIENT vigem;
static PVIGEM_TARGET target;
static XUSB_REPORT report;

int main(int argc, char** argv) {
	vigem = vigem_alloc();
	ASSERT(vigem != NULL);
	ASSERT(vigem_connect(vigem) == VIGEM_ERROR_NONE);
	XUSB_REPORT_INIT(&report);
	target = vigem_target_x360_alloc();
	ASSERT(vigem_target_add(vigem, target) == VIGEM_ERROR_NONE);
	
	while (1) {
		report.sThumbLX ++;
		ASSERT(vigem_target_x360_update(vigem, target, report) == VIGEM_ERROR_NONE);
		Sleep(10);
	}
}
```

... compiled with g++ from MinGW package.

When code reaches `vigem_target_add`, it blocks briefly and I can clearly hear "ding" from new device notification. Right after it, entire system freezes, often even with last audio sample looping indefinitely.

When ran on Windows 10 (recompiled or even same executable) or when DS4 target is used, code works exactly as expected. 

**Expected behavior**
Not freezing.

**Screenshots**
I've captured output from serial console and [uploaded it to pastebin](https://pastebin.com/zyPzYsgJ), just in case it will be somehow useful. I believe it's same message repeated.

**System details (please complete the following information):**
 - OS: Windows 7
 - Everything 32bit
 - Driver Version: 01/19/2018,1.14.3.0
 - Tested both with Virtualbox and real hardware

nefarius, 2018-11-27:
Heya, thanks for the detailed report (although I don't get where you grabbed that serial output 😛 )

Stupid question: you do have the 32-Bit Microsoft Xbox 360 Controller driver installed, right?

nefarius, 2018-11-30:
Will try to replicate on one of my systems. Windows 7 32-Bit is on the bottom of the test bed 😉

nefarius, 2018-12-19:
Damn, my Windows 7 machine is a bastard, but I'm almost there 😅

nefarius, 2018-12-21:
Alright @kozec here we finally are, and it's not even Christmas yet 😅 

[image]

The latest, not yet released, build seems to work fine. It's a physical, patched up-to-date machine with Windows 7 32-Bit.

Just to be sure I'll try your snippet and the older revision as well and report.

nefarius, 2018-12-21:
Hm, wasn't able to provoke it with version 1.14.3.0 as well 🤔

kozec, 2018-12-21:
I've tried to replicate same screen and got up to the point where OS freezes again :(

[image]

Frankly, I don't really have any idea where to even start debugging this. Is that not yet released build available somewhere?

nefarius, 2018-12-21:
You could hop onto the Discord and we could establish a remote session 😉

kozec, 2018-12-21:
I was able to obtain minidumps from after system feezed.

I'm currently fighting with Visual Studio installer in order to analyze them, but [here they are](https://github.com/ViGEm/ViGEmBus/files/2704284/dumps.zip) in case they are somehow useful.

nefarius, 2018-12-22:
Thanks, two things:

- please be aware that if you publicly share memory dumps that you might accidentally leak confidential information stored in there like passwords or encryption-related data
- the dump reveals that there is `ScpVBus` installed which might be the culprit

nefarius, 2018-12-22:
And instead of a minidump a [kernel memory dump](https://www.datapath.co.uk/how-to-create-a-kernel-memory-dump-in-win7) is required for more details.

kozec, 2018-12-28:
Hello and sorry for late answer, I wanted to test this as much as possible (and attend xmas :))

You are right, uninstalling `ScpVBus` driver prevented OS from freezing, but now, if x360 target is used, W7 breaks in even more interesting way:

- code reaches vigem_target_add and stays there.
- New x360 device appears in device manager, but Windows complains that it can't load driver with "Code 31"
- entire process became unkillable by anything, from ^C to task manager
- Windows is no longer able to shut down, it just stays on ["Shutting down" screen](https://i.ytimg.com/vi/YsYMm0TEzXI/maxresdefault.jpg), but without freezing.

That's current status report, I'll try to reinstall x360 driver and then entire OS :)

nefarius, 2018-12-28:
Attending Christmas instead of tinkering with Windows? Understandable 😆 

Ah! That is good news (despite the system still getting rekt ofc. 😅 )

Depending on the version the `ScpVBus` driver has a serious flaw that can cause a kernel deadlock, so pretty much what you've originally seen.

What you are describing now is an "orphaned IRP request", IIRC I have a newer driver in the lab where this is fixed as well. Before you nuke the machine we could try that together in a remote session or I can guide you via Discord.

I recommend reinstalling the official Microsoft X360 drivers and retrying.

kozec, 2018-12-28:
So, after testing a lot: Installing OS from scratch with nothing but X360 drivers, mingw and vigem freezes, but uninstalling those drivers from W7 I've been testing it originally, rebooting and installing them back works, until I reboot. Then it freezes, but works after next reboot and keeps working so far. My guess is that it's related to that "installing new HW" dialog that popups when device is connected for 1st time.

I think I'll leave emulating DS4 pad by default for W7 and put all this in Wiki, so it's user, not me, who has to try not killing his own system :) If there is any other info I can provide, please ask, otherwise feel free to close this issue.

nefarius, 2018-12-29:
Hm, odd, but still a few hints I can go after, thank you!

Squall-Leonhart, 2019-04-17:
This is only known to occur when the xbox360 software for windows 7 is not installed.

nefarius, 2019-04-18:
@Squall-Leonhart right, that's also what I've experienced. Not sure if really still a bug or close-able...

nefarius, 2019-04-18:
Closed as not reproducible.

closed 2019-04-18 by nefarius