nefarius/ViGEmBus #13: Rumble state is not always sent to client code

Ryochan7, 2019-03-30:
**Describe the bug**

DS4Windows users have brought up the issue and I can reproduce it. There are times when a (0,0) feedback state is not sent to the client code so DS4Windows will replay old rumble data and the DS4 will constantly rumble. I did some digging using the trace messages and I have concluded that the issue is with the code that handles IOCTL_XUSB_REQUEST_NOTIFICATION requests in ViGEmBus.

https://github.com/Ryochan7/DS4Windows/issues/616

The major issue is that a game can send an updated rumble state while a previous rumble state update is still being processed by the client code. The ViGEmClient will then send the next notify request to the bus, that request will then be put into a queue and then have to wait until the game sends another rumble state change before the client code will retrieve an updated rumble state.

**To Reproduce**

1. Run Retroarch 1.7.6, use ParallelN64 core with rumble pack set for player 1, 
2. Load Legend of Zelda: Ocarina of Time ROM and run
3. Cause Link to get hurt. Roll into a fence in the first area
4. DS4 will constantly rumble until stopped in Options menu in DS4Windows or by disconnecting controller

**Expected behavior**

Controller will rumble briefly followed by having the controller rumble motors stop

**System details (please complete the following information):**

 - OS: Windows 10 x64
 - Feeder software: DS4Windows 1.7.2
 - Driver Version: 1.14.3.0

**Additional context**

I made a temporary fix but it might not be a good idea to implement as is; there might be a better approach. Instead of forwarding the notify request immediately to a queue, check if the current xusb device has different rumble data than the data retrieved with the request. If the data differs, process the data immediately. Here is an example diff.

```
diff --git a/sys/busenum.c b/sys/busenum.c
index 1f7df12..365a62f 100644
--- a/sys/busenum.c
+++ b/sys/busenum.c
@@ -493,9 +493,33 @@ NTSTATUS Bus_QueueNotification(WDFDEVICE Device, ULONG SerialNo, WDFREQUEST Requ
             break;
         }
 
-        status = WdfRequestForwardToIoQueue(Request, pdoData->PendingNotificationRequests);
-
-        break;
+		PXUSB_REQUEST_NOTIFICATION notify = NULL;
+
+		status = WdfRequestRetrieveOutputBuffer(Request, sizeof(XUSB_REQUEST_NOTIFICATION), (PVOID)&notify, NULL);
+
+		if (NT_SUCCESS(status) && (notify->LargeMotor != xusbData->Rumble[3] || notify->SmallMotor != xusbData->Rumble[4]))
+		{
+			// State changed since last request. Update now.
+			// Assign values to output buffer
+			notify->Size = sizeof(XUSB_REQUEST_NOTIFICATION);
+			notify->SerialNo = pdoData->SerialNo;
+			notify->LedNumber = xusbData->LedNumber;
+			notify->LargeMotor = xusbData->Rumble[3];
+			notify->SmallMotor = xusbData->Rumble[4];
+
+			TraceEvents(TRACE_LEVEL_VERBOSE,
+				TRACE_USBPDO,
+				"-- Test Rumble Buffer: %02X %02X",
+				notify->LargeMotor, notify->SmallMotor);
+
+			WdfRequestCompleteWithInformation(Request, status, notify->Size);
+		}
+		else
+		{
+			status = WdfRequestForwardToIoQueue(Request, pdoData->PendingNotificationRequests);
+		}
+
+		break;
     case DualShock4Wired:
 
         ds4Data = Ds4GetData(hChild);
```

mika-n, 2019-04-02:
I think that the a24657a commit doesn't fix this "stuck rumble motor" issue. Here a link to a post where I describe how I tested it via Dolphin emulator controller setup screen without using any game images (ie. easy to test and duplicate the issue). 

https://github.com/Ryochan7/DS4Windows/issues/642#issuecomment-479037308

nefarius, 2019-04-09:
@mika-n true, it was an attempt, we're still brainstorming. It needs to be addressed in the driver.

nefarius, 2019-04-18:
Closed as not a driver issue, verified that the events arrive in order in the driver as expected but the inverted calls in the user-mode library need proper synchronization. Will update the findings in the referenced issue of the client library.

closed 2019-04-18 by nefarius