nefarius/ViGEmBus #15: emulated controller not listed by usbview(usbTreeView) and usbTrace

walkongit, 2019-04-08:
This emulated usb controller is not listed when I use some usb tool, eg.
usbTreeView: www.uwe-sieber.de
usbTrace: http://sysnucleus.com/
It is nice If ViGemBus be recognized by them.
plus. If ViGem could be listed by vmware workstation or virtualbox. Then we could redirect emulated controller to vmware guest os(android, linux, macos...)

nefarius, 2019-04-18:
Good idea. Will probably need to implement a few other interfaces to make it represent itself as a "true" USB bus device.

atmgnd, 2019-08-26:
checkout https://sourceforge.net/p/usb-vhci/virtusb/ci/master/tree/

nefarius, 2020-05-23:
Progress 😎 

[image]

atmgnd, 2020-05-24:
Great. very nice work, can't wait to try.
Also can we make ViGEmBus a generic purpose emulated usb bus, like Microsoft USB Device Emulation(UDE)(I nerver used UDE, i dont like it). some like below:
├── Vuirtual Usb Bus (kernel mode)
│   └── Emulated Xbox Controller(Normal win32 exe)
│   └── Emulated Usb keyboard(Normal win32 exe)
│   └── Emulated Usb camera(Normal win32 exe)
│   └── Emulated xxx(Normal win32 exe)

here is another virtual usb bus (a leak comercial software) I found, I have checked that it is working.
https://github.com/seanxs/ung_sean

nefarius, 2020-08-24:
@atmgnd sorry I somehow missed that replay; I have exactly that in the pipeline, stay tuned 😄

nefarius, 2022-06-25:
This is a topic for the successor.

closed 2022-06-25 by nefarius