nefarius/ViGEmBus #20: Closing connection to ViGEmBus error

brothan323, 2019-07-14:
**Describe the bug**
I can connect the controller and it'll work for a while. Yet after a while the controller will loose connection and a message appears on the left side of the bottom bar, it reads "closing connection to ViGEmBus

**To Reproduce**
Steps to reproduce the behavior (example):
1. Start program 
2. Plug in device '....'
3. See error after a bit

**System details (please complete the following information):**
 - OS: Fresh install of Windows 7 Ultimate
 - Feeder software: DS4Windows
 - Driver Version: 1.7.9

**Additional context**
I've never had this issue before reinstalling Windows.

nefarius, 2019-07-15:
Driver version 1.7.9, what? That's none of my released version numbers...

brothan323, 2019-07-15:
Whoops. Wrong ds4windows.

closed 2019-07-15 by brothan323