nefarius/ViGEmBus #22: Driver deadlock

michael-fadely, 2019-08-10:
**Describe the bug**
With a virtual Xbox 360 controller enabled, programs that poll controller input will be unable to launch. They deadlock attempting to open a handle to the device. This happens with feeder applications as well.

**To Reproduce**
I've been unable to consistently reproduce it. It happens fairly frequently, but with no specific steps that I could identify. If applications begin failing to launch with a feeder and Xbox 360 endpoint, however:
1. Launch an application that polls for controller input (a game, Uplay, Steam, Battle.net).
2. The application will stall or fail to launch.
3. Attempt to disable the device with Device Manager/etc, and it will also stall.
4. When the feeder is killed, the Xbox 360 endpoint device will enter an error state, displayed in Device Manager.

**Expected behavior**
Some kind of error handling or error logging. ViGEmClient has error logging, but it doesn't appear to encounter any errors when this deadlock occurs.

**Screenshots**
[image]

I attempted to disable the device, Device Manager stalled, and both the Game Controllers dialog and Battle.net immediately launched. Opening another instance of Device Manager reveals that the device is still enabled:

[image]

**System details (please complete the following information):**
 - OS: Windows 10 Pro 1903
 - Feeder software: (https://github.com/michael-fadely/ds4wizard/)[ds4wizard]
 - Driver Version: 1.14.3.0

**Additional context**
It very well could be something I'm doing with my feeder, but I've been unable to figure it out.

nefarius, 2019-08-12:
Thanks for the report but could you please monitor the situation with the latest driver version as well? What you run is very outdated.

michael-fadely, 2019-08-12:
...Wow, that IS really old! I missed the latest stable release somehow. Will do!

michael-fadely, 2019-08-14:
Same issue with latest stable. I will do some experimenting to see if I can narrow down the issue.

MtDewFella, 2019-08-16:
I have been getting the same problem, wasn't sure if it was steam, glosc or vigem, I assume it's vigem since trying to disable it causes driver manager to freeze up, only fix I have found is to reboot the pc, oddly it never happens during play only after (not all the time only some times)

nefarius, 2019-08-19:
Dammit 😅

MtDewFella, 2019-08-20:
I completely uninstalled vigem and glosc, and reinstalled vigem first then glosc and so far I have had no issues, the last time I installed I just installed glosc like normal, which installed vigem (or was supposed to) but when I first used glosc that time I got the couldn't initialize vigem error, I then installed vigem without uninstalling anything, perhaps that caused the issue, I will keep and eye on it

nefarius, 2020-01-15:
Any news @michael-fadely ?

michael-fadely, 2020-01-16:
Still exhibiting the exact same behavior, unfortunately. Just having the virtual Xbox 360 device present when attempting to launch an application that uses XInput will *sometimes* cause this deadlock and require a hard reboot to allow any other physical devices to connect to, or disconnect from, the system.

mosihe, 2020-01-24:
> Still exhibiting the exact same behavior, unfortunately. Just having the virtual Xbox 360 device present when attempting to launch an application that uses XInput will _sometimes_ cause this [...]

I have the similar issue.
I want to know if your game program exits normally.
And How often does this issue occur.

michael-fadely, 2020-01-29:
Could this be due to my program's multi-threading? I am using [ViGEmClient](https://github.com/ViGEm/ViGEmClient). I call `vigem_alloc` and `vigem_connect` (as well as free and disconnect) on the main thread, but `vigem_target_add`, `vigem_target_x360_register_notification`, etc. are called on a secondary thread.

mosihe, 2020-01-30:
Hello, I also have the problem with a single thread.I now think that it should be a better way to reproduce the problem scene, although this is difficult.
In order to narrow the scope of the system environment, I want to know how often you will get an exception and whether the process will be forced kill.

mo.he@vrviu.com
 
From: Michael Fadely
Date: 2020-01-30 06:04
To: ViGEm/ViGEmBus
CC: mo.he-vrviu; Comment
Subject: Re: [ViGEm/ViGEmBus] Driver deadlock (#22)
Could this be due to my program's multi-threading? I am using ViGEmClient. I call vigem_alloc and vigem_connect (as well as free and disconnect) on the main thread, but vigem_target_add, vigem_target_x360_register_notification, etc. are called on a secondary thread.

michael-fadely, 2020-01-30:
In my program, everything appears to continue working. I simply unplug the DualShock 4 and it calls `vigem_remove_target` seemingly without error. I will implement some more robust error checking in this instance to be certain.

Corrodias, 2020-01-31:
This seems very similar to a problem I'm experiencing. i'm using DS4Windows, on Windows 10 1909. After about 3 days of uptime, Monster Hunter World will stall on its black startup screen and not proceed to the dev logo, but Special K's overlay is drawn, and Chrome can still start a new, GPU-accelerated video, so it's not a video driver failure. Attempting to "stop" and "start" DS4Windows may appear to work but results in a CPU-spinning thread. Attempting to reboot Windows causes it to hang on "Restarting" for several minutes, then BSOD.

I have **not** yet narrowed this down to being caused by ViGEmBus. It does seem like a likely culprit, especially after reading this thread. I'll try uninstalling it after the next time this happens. I will also attempt to get an xperf trace on the shutdown process that includes stacks when it's stuck.

Corrodias, 2020-01-31:
In fact, if it's going to do it again, today's the day. I have about 7 or 8 hours before I'm going to commence playing games, at which time I'll be doing the trace and so on if it did deadlock. If you have any advice or suggestions on helping diagnose this, providing them now would be timely.

nefarius, 2020-01-31:
Make sure kernel memory dumps are enabled and keep the dump file somewhere. It shouldn't be shared publicly but is the best start to get the culprit.

Corrodias, 2020-01-31:
Roger. Seems I'll have to reboot to enable that, so I'll get it ready for next time. I wouldn't get my hopes up, though; it didn't record the bugcheck event to the log the last time, despite the BSOD. (And it rebooted to BIOS setup instead of Windows for some reason; I was able to just exit with no changes and go right back to Windows.) I could only guess as to why. It wrote one of those "Dump file creation failed" events after booting up.

Since the last time it happened, I removed an old Logitech driver I had and replaced it with a newer one. If that was the actual culprit, I won't be seeing this situation again.

Corrodias, 2020-02-02:
It happened again today, so I took some traces. I'll link them to you in an email.

I launched MHW, and it froze on its black startup screen. I ran a trace and produced "out.etl". Upon unplugging the controller, DS4Windows reported losing connection to it, and MHW suddenly continued as normal. I was also able to exit and launch it again normally. Plugging the controller back in, DS4Windows did not log any new messages, such as connecting to it.

Attempting to restart got stuck, as before. I took a trace during that restart, and I hit reset after it was clear that it wasn't going anywhere, rather than let it play out for 10 minutes. Unfortunately, xbootmgr was not able to assemble a finalized trace file once Windows started up; it claimed it couldn't load something from the registry. Therefore, you have "kernel.etl", "shutdown_BASE+DIAG+LATENCY+PROFILE+INTERRUPT+DPC_1_um_premerge.etl", and "shutdown_BASE+DIAG+LATENCY+PROFILE+INTERRUPT+DPC_1_km_premerge.etl".

There's also the less-inclusive "shutdown_BASE+DIAG+LATENCY_1.etl" that is from a previous attempt at a restart trace during this stuck condition.

If it happens again in a few days, I now have kernel memory dumping enabled, so I'll force a bugcheck in the hopes that'll actually dump it, following Microsoft's instructions for doing so.

Corrodias, 2020-02-07:
The deadlock has not yet triggered again, but I'm still being vigilant and and ready to catch a memory dump when it does.

Corrodias, 2020-02-11:
Damn, failure. It finally happened again, and I initiated a crash, but Windows once again failed to write a dump file due to an unspecified error. It looks like the only other option for getting a kernel memory dump is enabling kernel debugging mode and dumping it with windbg. I'll try to get that set up for next time.

nefarius, 2020-02-11:
@Corrodias shame, but was it "random" or while a game accessed the controller? Because I was thinking about writing a small "fuzzing" application which may get us to the goal faster. Like rapid (dis-)connects, calling XInput APIs automated many many times etc. Might be with the try other than waiting and hoping 😅

Corrodias, 2020-02-12:
I am hesitant to enable kernel debugging for several days at a time because I'm wary of a performance impact, and Monster Hunter: World is already a very demanding game. Having a way of reproducing it that doesn't require waiting upward of 3 days would be excellent. I'd be eager to try such an application and wouldn't have a reason to avoid the debugging mode then.

I only notice it happening right as the game's initializing, while the window is black and Special K has drawn its banner, but before it displays the splash screens, most likely during xinput initialization. That said, I can't guarantee that the truth isn't that the state is being triggered before that and the game is merely _encountering_ it. I'm using DS4Windows with "exclusive mode" enabled, so the game cannot see the original DS4.

This may be important: my DS4 is actually connected through a third-party adapter called the Titan Two. That's connected to the PC by USB at all times, and I merely swap the controller between bluetooth and USB. It's supposed to perfectly emulate a DualShock 4 and by all appearances is successful in doing so. However, if it deviates in some small way, perhaps that could be triggering this issue? I could try connecting the controller directly by USB for a couple of weeks to see if that prevents the problem. However, if the Titan Two is misbehaving in some small way, I think it would benefit both its devs and you to understand how and why it's causing a problem, so one or both ends can mitigate it.

Thinking back, unfortunately, I see no pattern between things I do before a gaming session that works and things I do before one that doesn't. It's all the same stuff. I connect and disconnect my USB headset on occasion. I swap the controller between bluetooth and USB so it can charge.

Corrodias, 2020-02-13:
I have the XinputTest program from x360ce, so I can try opening and closing that 30 times under various conditions. I did mange to trigger it now but not on a second attempt. Maybe I have to have Steam open. Maybe I have to swap to bluetooth once first. I have to wait until after my pirate anime stream with friends before I can continue testing, but I'll see if I can get this thing to trigger manually.

Corrodias, 2020-02-13:
Things that don't trigger it:
Running XinputTest 60 times in a row, while Steam's running.
Running MHW 14 times.
Switching profiles in DS4Windows 20 times.
Running MHW while there are 15 instances of XinputTest open.

There really seems to be a time component to this, somehow. It just won't do it for me on demand.

Corrodias, 2020-02-25:
I gave in and enabled kernel debugging. Surprisingly, it seems to have no appreciable effect on performance, at least when the debugger isn't attached. Nice!

My plan is to run livekd64.exe and run a full dump from kernel mode in kd (with /f), while the system's running, the next time it happens. That'll be 32 GB, since that's how much RAM I have, and apparently when you tell it you want a "full" dump, you mean **full**.

Would you like me to use any other options instead?

Corrodias, 2020-03-11:
Starting to think that the game running in DX12 mode just doesn't trigger it, so I'm going to switch back to DX11 mode. Shrug emoji.

michael-fadely, 2020-03-11:
A combination that tends to have this issue a lot for me is uplay + Rainbow Six: Siege, if you've got that handy.

Corrodias, 2020-03-14:
@michael-fadely I don't, unfortunately. If you can reproduce it as well, then you could submit a memory dump.

I'm still waiting for it to happen, and I fear it won't with DX12 mode, but for unrelated reasons, I can't play in DX11 mode right now.

There are 2 ways:
1) The simple way, if it works. Configure Windows to do a kernel memory dump when it crashes, then manually crash it when the condition occurs. Instructions here: https://github.com/Ryochan7/DS4Windows/issues/1049#issuecomment-590027280 - The downside is that on some systems (including mine), Windows never writes memory dumps. Seems to be a common thing when you're booting from an SSD.
2) Install the Windows debugging kit, enable local kernel debugging with bcdedit, reboot, wait for the condition to occur, and use livekd to do a full memory dump, then send that to nefarius. This is a bit more work, but I have confirmed that there seems to be zero performance impact from enabling debugging. The main downsides here are uploading a gigantic dump file and giving nefarius access to anything secret you have in memory. I trust him with it.

AugmentedAntics, 2020-04-02:
Well, after several weeks of thinking I found a way around it, the deadlock started happening again. And this is the fastest record it's had, less than 1 day after the last reboot and it happens after a few hours in Dolphin.  I do have a memory dump after waiting for this to happen again, so if it helps, I can send it.

Corrodias, 2020-04-07:
Please do send it to nefarius. I assume the email address in their profile is accurate.

@nefarius 
I, too, have finally reproduced the issue, and will be sending an S3 download link. While MHW never triggered it again, restarting Steam did.

AugmentedAntics, 2020-04-07:
Alright, I sent it to him.

nefarius, 2020-04-07:
Received and will have a look, thanks for your involvement!

AugmentedAntics, 2020-04-07:
Did the file sharing work?

nefarius, 2020-04-07:
Yes, downloading right now!

Corrodias, 2020-04-16:
Just to clarify, mine is a live dump using livekd that was taken while Steam was in the state of trying to start but being stuck. I had not yet unplugged the controller.

krzesu0, 2020-04-21:
I've been having the same issue lately. Some days it would happen 3 to 4 times per day. I've changed to which port I connect the Steam Controller from USB 3.0 Hub to direct 2.0 port. No issues from then.
Using 1.16.116 with GloSC. _Maybe this will help somehow._

nefarius, 2020-05-08:
I'm back on the hunt 🤠

mosihe, 2020-05-08:
> 🤠

 🤠

nefarius, 2020-05-08:
I'll need test personnel once I baked something 🙃

mosihe, 2020-05-08:
> I'll need test personnel once I baked something 🙃

I am not very clear what you mean,
I can reproduce the deadlock every day,
And happy to help.

nefarius, 2020-05-08:
@mosihe I'd need people with the issue (like you) to put their system into testmode and run a new driver version for testing once I have binaries is what I mean 😅

mosihe, 2020-05-09:
> @mosihe I'd need people with the issue (like you) to put their system into testmode and run a new driver version for testing once I have binaries is what I mean 😅

ok, so where to download the new driver, or you mean compile the master driver version.

nefarius, 2020-05-09:
Hold your horses dude, I haven't even started yet, I'd announce it if there was anything to download and where from 😉

nefarius, 2020-05-13:
Ahoy! [Beta build with attempted fix is online here](https://buildbot.vigem.org/builds/ViGEmBus/feature/core/cpp-ify/1.17.241.0/bin/), fee free to try, just update it over the existing `Virtual Gamepad Emulation Bus` device in device manager. **Testsigned** so **testmode is mandatory!**

Cheers

AugmentedAntics, 2020-05-13:
So, what can be done if Windows nags that the "best driver is already installed"?

nefarius, 2020-05-13:
You force it. 

[image]

Navigate to the folder containing all the downloaded files, choose the single arch INF file and select that, acknowledge the warning dialog and you're good to go.

AugmentedAntics, 2020-05-14:
I have successfully installed the test driver, but unfortunately, the deadlocking is still an issue. As I'm typing this, having DS4Windows open while it is occurring stalls any applications looking for the controller, and I already know restarting my PC will also stall. I spent a few hours in Citra prior to it.

nefarius, 2020-05-14:
Are you a 100% sure that the beta driver is loaded, can you confirm with screenshots of device manager version please.

AugmentedAntics, 2020-05-14:
[image]

This is what it looks like for me.

nefarius, 2020-05-15:
*Sad noises* your view is correct. It happens while emulating x360 right?

AugmentedAntics, 2020-05-15:
Yes, that's right.

nefarius, 2020-05-15:
Meh, so the search continues.

nefarius, 2020-05-15:
@AugmentedAntics are you on the Discord btw.? Might need to change my investigation strategy and I need affected users since I can't replicate.

AugmentedAntics, 2020-05-15:
I am. I go by the same name, except there's a space between Augmented and Antics.

Corrodias, 2020-05-17:
I'm willing, but since the problem has become much rarer than it used to be for me (I've gone weeks of uptime without seeing it), and I have found no method of reproducing it, I can't guarantee that my participation would be useful. Corrodias#2162

nefarius, 2020-05-19:
[Here's the latest build with quite a few promising improvements.](https://buildbot.vigem.org/builds/ViGEmBus/feature/core/cpp-ify/1.17.258.0/disk1/)

nefarius, 2020-05-24:
Nope, still happens. Slowly but steadily running out of ideas.

gjmv, 2020-06-24:
Hi, i have a similar issue. 
I'm using GloSc and frequenly when i close the overlay, my device manager is frozed up, and the Steam Controller don't work anymore.
I have to reset the PC, it stucks in "rebooting", then i manually push reset button. (plus others problems like said in the thread)
Yesterday i removed glosc and vigembus driver, then install latest vigembus driver first, then install glosc, and today the issue occurs 2 times. (before this happened every 2 o 3 days, casuality?)
My system never crash (never BSOD) but needs a fisical reboot to get working again.
I forced a BSOD and send dump file to @nefarius 

This is an image of my device manager

[image]

Thanks!

nefarius, 2020-06-24:
Thanks @gjmv I could in the meantime sort of replicate it and I think the culprit is the ironically the mechanism to keep the plug-in request on hold until the device is ready. Although I haven't found a clear indicator of an issue within the logic itself, I have the suspicion that it somehow causes a race condition in conjunction with MS `xusb.sys` and stalls unrecoverable. I may need to redesign the entire plugin API but other than backwards compatibility that's necessary anyway, I don't like how it was designed in the first place.

michael-fadely, 2020-06-29:
How do you currently replicate it?

I've run into this issue pretty persistently with Dark Souls 3 and would be willing to do some debugging of my own. Tangentially because I'm experiencing input lag in this game using my feeder, but that might be my fault. 😄

nefarius, 2020-06-30:
Instead of [doing this](https://github.com/ViGEm/ViGEmBus/blob/9cbd4b65c2f9e017cd3918f3191341e80ee1df30/sys/busenum.cpp#L222-L224), just return `STATUS_SUCCESS` basically.

[38 more replies]

closed 2022-08-08 by nefarius