nefarius/ViGEmBus #24: Driver Error

pokeygithub, 2019-11-30:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior (example):
1. Start program '...'
2. Click on '....'
3. Plug in device '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System details (please complete the following information):**
 - OS: [e.g. Windows 10 1803]
 - Feeder software: [e.g. VDX, DS4Windows, ...]
 - Driver Version: [e.g. 1.14.3.0]

**Additional context**
Add any other context about the problem here.

pokeygithub, 2019-11-30:
Windows 10 Version 1903 fully patched.  PS4 Controller shows that it is connected, then a few seconds later it shows Driver error.  Click PS4 controller connects again.  few seconds later Driver error.  This is with version 1.16.115.

pokeygithub, 2019-11-30:
It does work correctly when plugged in via USB, Bluetooth is what is failing for me.

nefarius, 2019-12-08:
Could you provide some screenshots of the areas where said error occurs? Because from the description alone I don't quite get what the issue is about, thanks!

pokeygithub, 2019-12-11:
I am so sorry! I forgot to update this ticket.  Ended up the brand new USB to BT/Wifi Adapter just quit sending BT.  Replaced it with a new Killer BT/Wifi adapter and has been working perfectly ever since!  I really appreciate everything you have done with this driver as using PS4 controllers on my PC is an absolute dream!  Once again thank you for your hard work and willingness to help this dumb old guy.

closed 2019-12-11 by pokeygithub