nefarius/ViGEmBus #25: Setup isn't working

bolaum, 2019-12-04:
**Describe the bug**
I downloaded the latest installer (ViGEmBus_Setup_1.16.115.exe) and tried to run it. Absolutely nothing happens. I already tried as admin and through powershell, nothing is printed either. I found out that GloSC wasn't working because the driver wasn't installed, so I tried to install it manually.

**To Reproduce**
1. Download the setup file
2. Try to run it (double-click)
3. Nothing opens.

**Expected behavior**
The driver to be installed.

**System details (please complete the following information):**
 - OS: Windows 10 Pro 1909
 - Feeder software: GloSC 
 - Driver Version: 1.16.115

Are there any flags I can pass to the installer to make it verbose and give me any clues on what is happening?

nefarius, 2019-12-08:
Hello!

Please try to run it from the command line with the following argument: `/exenoupdates`

See if that helps, cheers!

bolaum, 2019-12-08:
Powershell x64 as admin:
```
PS C:\users\bolaum\Downloads> .\ViGEmBus_Setup_1.16.115.exe /exenoupdates
PS C:\users\bolaum\Downloads>
```
Still nothing happens =/

There's no `verbose` flag? 

If I build it locally, will it generate the installer binary? Will the unsigned version work the same way?

nefarius, 2019-12-08:
You need Advanced Installer with an Enterprise license to create the setup, so the build won't provide it for you and driver signing has no relevance to the setup not starting. [You could also try this](https://forums.vigem.org/topic/290/troubleshooting-a-failed-vigem-bus-setup).

bolaum, 2019-12-08:
This is so weird, no log is generated. It is as if the executable was never run.
```
PS C:\users\bolaum\Downloads\tmp> .\ViGEmBus_Setup_1.16.115.exe /L*V .\install.log
PS C:\users\bolaum\Downloads\tmp> ls

    Directory: C:\users\bolaum\Downloads\tmp

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
-a----        12/3/2019  10:46 PM       14310704 ViGEmBus_Setup_1.16.115.exe

PS C:\users\bolaum\Downloads\tmp>
```
Same result with `/exenoupdates`. I'm a programmer, but Windows isn't really my thing. 

About the signing, I wasn't implying it was the problem. I just wanted to know if Windows will allow an unsigned version to be installed, but it seems you generate it with an external tool.

bolaum, 2019-12-08:
Is there something equivalent do `strace` or `ptrace` for Windows / PS?

nefarius, 2019-12-08:
[Do any of these work?](https://www.advancedinstaller.com/user-guide/exe-setup-file.html) Is there anything generated in Windows Event Log when you try to run it?

bolaum, 2019-12-08:
Nothing in the Event Viewer. Same results with other params.

I ran the executable with IDA and it seems to be breaking after loading `SysWOW64\ntdll.dll`:
```
WOW64 process has been detected (pid=5652)
E0000: process C:\Users\bolaum\Downloads\tmp\ViGEmBus_Setup_1.16.115.exe has started (pid=5652)
2B8000: created segment ViGEmBus_Setup_1.16.115.exe, end: 2F3000
300000: created segment debug001, end: 320000
320000: created segment debug002, end: 323000
330000: created segment debug003, end: 34B000
385000: created segment debug004, end: 388000
388000: created segment debug005, end: 390000
390000: created segment debug006, end: 394000
3A0000: created segment debug007, end: 3A3000
3B0000: created segment debug008, end: 3B2000
498000: created segment TIB[0000097C], end: 49D000
6FD000: created segment debug009, end: 6FF000
6FF000: created segment debug010, end: 700000
77520000: created segment ntdll.dll, end: 77521000
77521000: created segment ntdll.dll, end: 7763B000
7763B000: created segment ntdll.dll, end: 77641000
77641000: created segment ntdll.dll, end: 77642000
77642000: created segment ntdll.dll, end: 77644000
77644000: created segment ntdll.dll, end: 776BA000
7FFE0000: created segment debug011, end: 7FFE1000
7FFE8000: created segment debug012, end: 7FFE9000
FF740000: created segment debug013, end: FF741000
FF750000: created segment debug014, end: FF773000
7FFC250E0000: created segment ntdll.dll, end: 7FFC250E1000
7FFC250E1000: created segment ntdll.dll, end: 7FFC251F8000
7FFC251F8000: created segment ntdll.dll, end: 7FFC2523F000
7FFC2523F000: created segment ntdll.dll, end: 7FFC25245000
7FFC25245000: created segment ntdll.dll, end: 7FFC25246000
7FFC25246000: created segment ntdll.dll, end: 7FFC2524B000
7FFC2524B000: created segment ntdll.dll, end: 7FFC2525A000
7FFC2525A000: created segment ntdll.dll, end: 7FFC2525B000
7FFC2525B000: created segment ntdll.dll, end: 7FFC2525E000
7FFC2525E000: created segment ntdll.dll, end: 7FFC252D0000
7FFC250E0000: loaded C:\Windows\System32\ntdll.dll
7763E000: created segment ntdll.dll, end: 77641000
7763B000: updated segment ntdll.dll, end: 7763E000
7763F000: created segment ntdll.dll, end: 77641000
7763E000: updated segment ntdll.dll, end: 7763F000
77520000: loaded C:\Windows\SysWOW64\ntdll.dll
PDBSRC: loading symbols for 'C:\Windows\SysWOW64\ntdll.dll'...
PDB: using load address 77520000
Debugger: process has exited (exit code 0)
```
Is the installer x32?

nefarius, 2019-12-08:
Very odd. The installer is multi-arch, the setup is just a bootstrapper around the actual embedded MSIs... Since it's not my product (Advanced Installer) I could try to get support involved, IDK at this stage what to recommend honestly...

bolaum, 2019-12-08:
No worries! Could you make the install scripts and raw msi files available?

Since you are paying for it, you sure are entitled for some support =D maybe they've seen this elsewhere. I can provide all info you need.

nefarius, 2019-12-09:
We could experiment with alternative builds of the setup together "in private" on my Discord for example, usually it has worked best in the past if those issues are tracked down together, maybe also involving a remote session. There's been a couple of oddities in the past I simply could never replicate on my machines 😁

uetapoh, 2019-12-18:
Hello. I just had this problem here. After formatting my PC  the installation was successful.
Using Windows 8.1 here.

nefarius, 2019-12-18:
> Hello. I just had this problem here. After formatting my PC the installation was successful.
> Using Windows 8.1 here.

Soooo... You nuked your Windows installation which had the problem and now with a fresh one you don't have the problem anymore? Thanks for sharing but that effectively nuked any chance to properly diagnose and fix as well 🤷‍♀️

uetapoh, 2019-12-18:
Sorry. I remember that I used the PS3 controler with the ScpDriver here. Maybe that was the problem.

nefarius, 2019-12-18:
Very unlikely, at this stage of the setup no driver shenanigans is involved at all.

AsuraMazda, 2019-12-24:
Hi, got the same problem here... I managed to generate a logfile (combining the log command AND the exenoupdates), but i have no idea what anything of that means, gonna upload the log file, hope it helps. 
Trying to install to use with DS4.

OS: Windows 10 Pro 1903 x64
Driver Version: 1.16.115

[install.log](https://github.com/ViGEm/ViGEmBus/files/3998969/install.log)

henriquel1997, 2019-12-28:
Hi, i have the same issue but i wasn't able to generate a log file through the command line.
Is there anything else i could do to help solve this?

nefarius, 2019-12-28:
*grumpy Advanced Installer noises*

henriquel1997, 2019-12-28:
There's not much i can do i guess?

nefarius, 2019-12-28:
***Stay tuned...***

[image]

megadrago88, 2019-12-28:
Hehe

nefarius, 2019-12-28:
@bolaum @AsuraMazda @henriquel1997 [I've published a pre-release addressing the issues](../releases/tag/setup-v1.16.116), please test and report, thank you!

henriquel1997, 2019-12-28:
Seems to be working fine here!

bolaum, 2019-12-29:
@nefarius wow! Much installer, very working!

Dude, please tell me what you did. I'm professionally curious =D

nefarius, 2019-12-29:
@bolaum nothing too exciting I'm afraid, I removed the pre-setup update check and re-built the setup with latest Advanced Installer and there we have it 😅

nefarius, 2020-01-04:
Alright, I think we are good to go 😄

closed 2020-01-04 by nefarius

HughWish, 2020-02-18:
This is still a problem.

OS Name:                   Microsoft Windows 10 Home
OS Version:                10.0.18363 N/A Build 18363

ViGEmBus_Setup_1.16.116.exe /exenoupdates

This page in the forums is full of dead links and the repository does not seem to work.
https://forums.vigem.org/topic/39/new-vigem-installer

nefarius, 2020-02-18:
@HughWish the installation methods referenced on that post don't exist anymore for like a year now.