nefarius/ViGEmBus #27: What is the mistake?

Arongames, 2020-01-07:
Assertion failed: Could not find 'git' via 'C:\Windows\System32\where.exe'.
   at Nuke.Common.ControlFlow.Fail(String text)
   at Nuke.Common.ControlFlow.NotNull[T](T obj, String text)
   at Nuke.Common.Tooling.ToolPathResolver.GetPathExecutable(String pathExecutable)
   at Nuke.Common.Tools.Git.GitTasks.GitCurrentBranch(String workingDirectory)
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at Nuke.Common.Git.GitRepositoryAttribute..ctor()
   at System.RuntimeTypeHandle.CreateCaInstance(RuntimeType type, IRuntimeMethodInfo ctor)
   at System.Reflection.CustomAttribute.GetCustomAttributes(RuntimeModule decoratedModule, Int32 decoratedMetadataToken, Int32 pcaCount, RuntimeType attributeFilterType, Boolean mustBeInheritable, IList derivedAttributes)
   at System.Reflection.CustomAttribute.GetCustomAttributes(RuntimeFieldInfo field, RuntimeType caType)
   at System.Attribute.GetCustomAttributes(MemberInfo element, Type type, Boolean inherit)
   at System.Reflection.CustomAttributeExtensions.GetCustomAttributes[T](MemberInfo element)
   at Nuke.Common.Execution.BuildExtensions.<>c.<GetInjectionMembers>b__4_0(MemberInfo x)
   at System.Linq.Enumerable.WhereArrayIterator`1.ToList()
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at Nuke.Common.Execution.BuildExtensions.GetInjectionMembers(NukeBuild build)
   at Nuke.Common.Execution.InjectionService.InjectValues(NukeBuild build)
   at Nuke.Common.Execution.BuildExecutor.Execute[T](Expression`1 defaultTargetExpression)

Repeating warnings and errors:
Assertion failed: Could not find 'git' via 'C:\Windows\System32\where.exe'.

nefarius, 2020-01-07:
Wow. This has to be the laziest issue I've come across in a long time. I'm lost for words, honestly.

Arongames, 2020-01-07:
I am stupid little bitсh, pls help)))

nefarius, 2020-01-07:
Come on, mate, don't do this to us. So I assume you try to build and it fails you? How? Why? Just build within the Visual Studio GUI then?

Arongames, 2020-01-07:
I can’t understand why, when running the PowerShell script, it gives this error

Evengard, 2020-01-09:
It's written in english: "I didn't found GIT" Install GIT, man.

closed 2020-01-15 by nefarius