nefarius/ViGEmBus #28: ViGEm Updates Has Expired SSL Certificate

quinn-brittain, 2020-01-12:
The certificate expired on the 10th and hasn't been updated.
I noticed this because I am now getting a warning window on Windows10.
[image]
[image]
[image]

nefarius, 2020-01-12:
Never have I received the same message on so many different channels 😅 and ofc. this had to happen the one time I'm traveling without PC/Laptop... Will fix ASAP 😜

nefarius, 2020-01-12:
Fixed 😎

closed 2020-01-12 by nefarius