nefarius/ViGEmBus #31: Stadia Controller Support

Syberiyxx, 2020-02-14:
**Is your feature request related to a problem? Please describe.**
I would like to use a Stadia Controller with Parsec but it is misidentified. Buttons are registering as triggers and the joysticks are double mapped to the triggers and joysticks simultaneously. 

**Describe the solution you'd like**
Stadia Controller support.

**Describe alternatives you've considered**
I'm attempting to use SDL2 Gamepad Tool to interpret the Stadia Controller for Parsec

**Additional context**
none

nefarius, 2020-02-14:
What exactly do you mean by "Stadia Controller support"? That the controller should be added as a possible emulated **output** device? Because for an **input** ViGEm has an open general-purpose API, there is no need for it to provide anything Stadia-specific.

Syberiyxx, 2020-02-14:
Hmm, I'm having a hard time getting to the bottom of where the issue is. I have a Stadia Controller and in Parsec it shows as a generic controller but the mappings are wrong. It's registering an analog trigger as a button, etc etc. I misunderstood what ViGEm did as an emulator, not an interpreter. Sorry for the misunderstanding

nefarius, 2020-02-14:
Happens, mate 😁 ViGEm spits out what it gets fed so I'd advise you inform the official Parsec support about this 👍

closed 2020-02-14 by nefarius