nefarius/ViGEmBus #32: Possibly add Touchmote to Known users of ViGEm list

Ryochan7, 2020-02-14:
To preface this, Touchmote is a Wiimote mapper utility that can be used to navigate the Windows desktop with a Wiimote and for playing games with the Wiimote by emulating an Xbox 360 controller. Although the official Touchmote project is pretty much dead, I have forked it and I have been updating it for about two months. One of the big updates was to rewrite the XInput output handler to use ViGEmBus rather than ScpVBus.

Official site:
http://touchmote.net/

Official repo:
https://github.com/simphax/Touchmote

Forked version repo:
https://github.com/Ryochan7/Touchmote/tree/ryochan7

I have only one old example of using Xbox 360 controller emulation in Touchmote; my main experiments have been improving the FPS mouse routine. There is a short video showing Star Wars Episode 1: Racer being played with a Wiimote. The Wiimote was placed in a Wheel shell to mimic playing the game with steering wheel. Although updates to Touchmote would make this type of configuration better now, the game was pretty fun playing this way and I played through the whole campaign using the Wiimote.

https://www.youtube.com/watch?v=XgwFgKQwczk

As an aside, HidGuardian was needed to get controller support working properly in the game. The game initially detected the inactive DolphinBar created controllers and would refuse to detect the emulated Xbox 360 controller. Hiding the 4 DolphinBar created controllers was required for the game to detect the virtual Xbox 360 controller.

nefarius, 2020-05-10:
Thanks for bringing it to my attention, looks very interesting, I haven't forgotten about it, just trying to prioritize 😅

nefarius, 2020-05-22:
Done

closed 2020-05-22 by nefarius

Trihy, 2022-10-10:
Hi, there is any easy way to compile this fow windows? Or where to find compiled file. There was also a bat file to sign touchmote. 

I just want to get rid of touchmote emulation driver and use vigem.