nefarius/ViGEmBus #33: Less intrusive updater notifications

mbc07, 2020-03-08:
**Is your feature request related to a problem? Please describe.**
Kinda. The ViGEm Updater is completely silent for most of the time, however, when an update is available, it is intrusive since it immediately opens a window on top of everything else asking if you want to update. I've also noticed the updater window will even kick you off of any running fullscreen application (similar to the Windows Firewall notification when it blocks something) if the update check happens while a fullscreen app or game is running. 

**Describe the solution you'd like**
I have no issues with the actual updater window, in fact its wizard-based approach is very concise and direct to the point. However, I do think it shouldn't immediately open on top of everything else whenever an update is available. 

I think displaying a system tray icon and pushing a regular notification when an update is available  (e.g. similar to the behavior of safely removing an USB device), then bringing the updater window only upon clicking on the system tray icon or on the notification is way less intrusive, while still retaining the current update flow.

**Describe alternatives you've considered**
I considered disabling the updater and manually checking for new versions, but I ended leaving it enabled despite the current intrusive behavior, as in the end it still is way more practical than manually checking for new versions.

nefarius, 2020-03-25:
The updater is a 3rd party component from Advanced Installer I only have so much influence over, last time I checked there was no tray application feature.

I'll see if in the meantime some new options have been introduced.

nefarius, 2020-04-18:
I can't influence this as of right now and updates are fairly spars anyway so in reality not that big of a deal.

closed 2020-04-18 by nefarius