nefarius/ViGEmBus #36: Installer "Invalied drive Z:"

lividus, 2020-03-25:
**Describe the bug**
Installer (1.16.116) show warninng window "Invalied drive Z:" and installation is interrupted

To Reproduce
Just start installer and click install

System:
 - OS: 10 1806
 - Driver Version: 1.16.116

**Additional context**
Drive Z: is webdav

nefarius, 2020-03-25:
You can not run an elevated application from a network drive, that's a Windows limitation, nothing I can help you with.

lividus, 2020-03-25:
> 
> 
> You can not run an elevated application from a network drive, that's a Windows limitation, nothing I can help you with.

The network drive is simply present in the system. Z: not use for start installer or install to them. If i disable all network drives installer work as expeted.

nefarius, 2020-03-25:
That's quite an important detail 😆

nefarius, 2020-12-22:
Can't reproduce on v1.17.333

closed 2020-12-22 by nefarius