nefarius/ViGEmBus #37: An unexpected error has occurred

johnnygodsa, 2020-03-26:
**Describe the bug**

I downloaded the latest installer (1.16.116) and tried to run it.
but this error always occurs

> An unexpected error has occurred. please run the setup program again.
> if problems continue, see the microsoft hardware web site for software updates and additional product information to help you troubleshoot the issue.

**To Reproduce**

1. Download the setup file
2. Try to run it 

**System details** 

Windows 10 pro 1903 (1836.2.720)

Evengard, 2020-04-29:
Why not use WiX toolset for the installer creation? I guess it would solve most of the installer based problems, and remove a dependency on a closed-source software?

nefarius, 2020-04-29:
More and more tempted to do after wondering why such a fairly simple project appears to cause such headaches 🤔 Advanced Installer is awfully more convenient than WiX though, hands down.

Evengard, 2020-04-30:
There is also Inno Setup though. Not as advanced as WiX but maybe this project doesn't require anything more sophisticated...

nefarius, 2020-12-21:
Setup has been WiX-ified 😅

closed 2020-12-21 by nefarius