nefarius/ViGEmBus #38: Release notes are confusing

jpflouret, 2020-04-06:
**Describe the bug**
The release notes start with "Necessary preparations for Windows 7" and finish with "These steps are not necessary for Windows 8.1 or 10!". There are no clear install instructions for windows 8.1 or 10 and some users have skipped installing the drivers due to the ambiguity of the description.

**To Reproduce**
Steps to reproduce the behavior (example):
1. Navigate to [the releases page](https://github.com/ViGEm/ViGEmBus/releases/tag/setup-v1.16.116).
1. Observe release notes.

**Expected behavior**
A clear and concise description on how to install ViGEmBus drivers for Windows 8.1 and 10 is included in addition to the Windows 7 instructions.

**Screenshots**
Not applicable.

**System details (please complete the following information):**
 - OS: Windows 10
 - Feeder software: [RdpGamepad](https://github.com/microsoft/RdpGamepad)
 - Driver Version: 1.16.116

**Additional context**
None

Evengard, 2020-04-07:
Actually for Windows 8-10 installation you just run the installer... And that's all.

nefarius, 2020-04-07:
I have to agree, prerequisites aren't really release notes and should be moved somewhere else. Wiki maybe?

> There are no clear install instructions for windows 8.1 or 10 and some users have skipped installing the drivers due to the ambiguity of the description.

... I digress 😆 will come up with a new *user-friendly* scripture 😬

nefarius, 2020-04-07:
And done. Let me know what you think.

jpflouret, 2020-04-07:
That looks great! Thanks.

closed 2020-04-07 by jpflouret