nefarius/ViGEmBus #39: [Cannot Build] inf2cat fails

Davidobot, 2020-04-19:
I'm trying to build the driver from source, but the inf2cat stage keeps failing.

`Errors:
22.9.1: vigembus.sys in [drivers_dir] of ViGEmBus\vigembus.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
22.9.1: wdfcoinstaller01009.dll in [vigembus_device_coinstaller_copyfiles] of ViGEmBus\vigembus.inf is missing or cannot be decompressed from source media.  Please verify all path values specified in SourceDisksNames, SouceDisksFiles, and CopyFiles sections resolve to the actual location of the file, and are expressed in terms relative to the location of the inf.
`

I've tried next to everything -- playing around with the WDK versions and MSVC versions, but nothing seems to help. Any advice?

Davidobot, 2020-04-19:
Fixed by changing
```
[SourceDisksNames.amd64]
1 = %DiskName%,,,"\x64"
[SourceDisksNames.x86]
1 = %DiskName%,,,"\x86"
```
to
```
[SourceDisksNames]
1 = %DiskName%,,,
```

Still, an explanation would be nice as to why there is a broken .inf in the project?

nefarius, 2020-04-20:
The .inf isn't "broken", it's designed for production releases where one .inf for binaries of both architectures exist within these sub-directories. Just disable inf2cat in VS and run it manually when the folder structure has been set up that way. Or use it the way you adapted it, works as well.

closed 2020-04-20 by nefarius