nefarius/ViGEmBus #40: Xbox One HID Report Descriptior

DJm00n, 2020-04-25:
While looking at original `xboxgip.sys` (version 10.0.18362.267) and how it works I found `gipControllerHidDescriptor` var that this driver uses for its HID Physical Device Object. It was [`HID_DESCRIPTOR`](https://docs.microsoft.com/windows-hardware/drivers/ddi/hidport/ns-hidport-_hid_descriptor) struct.
Stripped header and decoded it via [RDD! HID Report Descriptor Decoder](https://github.com/abend0c1/hidrdd) tool. Maybe could be userfull in this or other projects:
```cpp

//--------------------------------------------------------------------------------
// Decoded Application Collection
//--------------------------------------------------------------------------------

/*
05 01        (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page 
09 05        (LOCAL)  USAGE              0x00010005 Game Pad (Application Collection)  
A1 01        (MAIN)   COLLECTION         0x01 Application (Usage=0x00010005: Page=Generic Desktop Page, Usage=Game Pad, Type=Application Collection)
A1 20          (MAIN)   COLLECTION         0x20 Reserved
09 30            (LOCAL)  USAGE              0x00010030 X (Dynamic Value)  
09 31            (LOCAL)  USAGE              0x00010031 Y (Dynamic Value)  
15 20            (GLOBAL) LOGICAL_MINIMUM    0x20 (32)  
27 FFFF2020      (GLOBAL) LOGICAL_MAXIMUM    0x2020FFFF (539033599)  
95 02            (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields  
75 10            (GLOBAL) REPORT_SIZE        0x10 (16) Number of bits per field  
81 02            (MAIN)   INPUT              0x00000002 (2 fields x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap  <-- Error: REPORT_SIZE (16) is too small for LOGICAL_MAXIMUM (539033599) which needs 30 bits.
C0             (MAIN)   END_COLLECTION      
A1 20          (MAIN)   COLLECTION         0x20 Reserved
09 33            (LOCAL)  USAGE              0x00010033 Rx (Dynamic Value)  
09 34            (LOCAL)  USAGE              0x00010034 Ry (Dynamic Value)  
15 20            (GLOBAL) LOGICAL_MINIMUM    0x20 (32) <-- Redundant: LOGICAL_MINIMUM is already 32 
27 FFFF2020      (GLOBAL) LOGICAL_MAXIMUM    0x2020FFFF (539033599) <-- Redundant: LOGICAL_MAXIMUM is already 539033599 
95 02            (GLOBAL) REPORT_COUNT       0x02 (2) Number of fields <-- Redundant: REPORT_COUNT is already 2 
75 10            (GLOBAL) REPORT_SIZE        0x10 (16) Number of bits per field <-- Redundant: REPORT_SIZE is already 16 
81 02            (MAIN)   INPUT              0x00000002 (2 fields x 16 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap  <-- Error: REPORT_SIZE (16) is too small for LOGICAL_MAXIMUM (539033599) which needs 30 bits.
C0             (MAIN)   END_COLLECTION      
05 01          (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page <-- Redundant: USAGE_PAGE is already 0x0001
09 32          (LOCAL)  USAGE              0x00010032 Z (Dynamic Value)  
15 20          (GLOBAL) LOGICAL_MINIMUM    0x20 (32) <-- Redundant: LOGICAL_MINIMUM is already 32 
26 FF03        (GLOBAL) LOGICAL_MAXIMUM    0x03FF (1023)  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields  
75 0A          (GLOBAL) REPORT_SIZE        0x0A (10) Number of bits per field  
81 02          (MAIN)   INPUT              0x00000002 (1 field x 10 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
15 20          (GLOBAL) LOGICAL_MINIMUM    0x20 (32) <-- Redundant: LOGICAL_MINIMUM is already 32 
25 20          (GLOBAL) LOGICAL_MAXIMUM    0x20 (32)  
75 06          (GLOBAL) REPORT_SIZE        0x06 (6) Number of bits per field  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields <-- Redundant: REPORT_COUNT is already 1 
81 03          (MAIN)   INPUT              0x00000003 (1 field x 6 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
05 01          (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page <-- Redundant: USAGE_PAGE is already 0x0001
09 35          (LOCAL)  USAGE              0x00010035 Rz (Dynamic Value)  
15 20          (GLOBAL) LOGICAL_MINIMUM    0x20 (32) <-- Redundant: LOGICAL_MINIMUM is already 32 
26 FF03        (GLOBAL) LOGICAL_MAXIMUM    0x03FF (1023)  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields <-- Redundant: REPORT_COUNT is already 1 
75 0A          (GLOBAL) REPORT_SIZE        0x0A (10) Number of bits per field  
81 02          (MAIN)   INPUT              0x00000002 (1 field x 10 bits) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
15 20          (GLOBAL) LOGICAL_MINIMUM    0x20 (32) <-- Redundant: LOGICAL_MINIMUM is already 32 
25 20          (GLOBAL) LOGICAL_MAXIMUM    0x20 (32)  
75 06          (GLOBAL) REPORT_SIZE        0x06 (6) Number of bits per field  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields <-- Redundant: REPORT_COUNT is already 1 
81 03          (MAIN)   INPUT              0x00000003 (1 field x 6 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
05 09          (GLOBAL) USAGE_PAGE         0x0009 Button Page 
19 01          (LOCAL)  USAGE_MINIMUM      0x00090001 Button 1 Primary/trigger (Selector, On/Off Control, Momentary Control, or One Shot Control)  
29 0A          (LOCAL)  USAGE_MAXIMUM      0x0009000A Button 10 (Selector, On/Off Control, Momentary Control, or One Shot Control)  
95 0A          (GLOBAL) REPORT_COUNT       0x0A (10) Number of fields  
75 01          (GLOBAL) REPORT_SIZE        0x01 (1) Number of bits per field  
81 02          (MAIN)   INPUT              0x00000002 (10 fields x 1 bit) 0=Data 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap  <-- Error: REPORT_SIZE (1) is too small for LOGICAL_MINIMUM (32) which needs 6 bits. <-- Error: REPORT_SIZE (1) is too small for LOGICAL_MAXIMUM (32) which needs 6 bits.
15 20          (GLOBAL) LOGICAL_MINIMUM    0x20 (32) <-- Redundant: LOGICAL_MINIMUM is already 32 
25 20          (GLOBAL) LOGICAL_MAXIMUM    0x20 (32) <-- Redundant: LOGICAL_MAXIMUM is already 32 
75 06          (GLOBAL) REPORT_SIZE        0x06 (6) Number of bits per field  
95 01          (GLOBAL) REPORT_COUNT       0x01 (1) Number of fields  
81 03          (MAIN)   INPUT              0x00000003 (1 field x 6 bits) 1=Constant 1=Variable 0=Absolute 0=NoWrap 0=Linear 0=PrefState 0=NoNull 0=NonVolatile 0=Bitmap 
05 01          (GLOBAL) USAGE_PAGE         0x0001 Generic Desktop Page 
09 39          (LOCAL)  USAGE              0x00010039 Hat switch (Dynamic Value)  
15 01          (GLOBAL) LOGICAL_MINIMUM    0x01 (1)  
25 08          (GLOBAL) LOGICAL_MAXIMUM    0x08 (8)  
35 20          (GLOBAL) PHYSICAL_MINIMUM   0x20 (32)  
[... 96 lines cut ...]
  uint8_t  GD_GamePadHatSwitch : 4;                  // Usage 0x00010039: Hat switch, Value = 1 to 8, Physical = ((Value - 1) x 283 / 7 + 32) in 
  uint8_t  : 4;                                      // Pad
                                                     // Collection: CA:GamePad :SystemControl
  uint8_t  GD_GamePadSystemControlSystemMainMenu : 1; // Usage 0x00010085: System Main Menu, Value = 32 to 1, Physical = ((Value - 32) x 0 / -31 + 32) in 10¹⁶  units
  uint8_t  : 7;                                      // Pad
                                                     // Collection: CA:GamePad
  uint8_t  GEN_GamePadBatteryStrength;               // Usage 0x00060020: Battery Strength, Value = 32 to 8447, Physical = (0 + 32) in 10¹⁶  units
} inputReport_t;

//--------------------------------------------------------------------------------
// Physical Interface Device Page outputReport (Device <-- Host)
//--------------------------------------------------------------------------------

typedef struct
{
                                                     // No REPORT ID byte
                                                     // Collection: CA:GamePad CL:
  uint8_t  PID_GamePadDcEnableActuators : 4;         // Usage 0x000F0097: DC Enable Actuators, Value = 32 to 1, Physical = ((Value - 32) x 0 / -31 + 32) in 
  uint8_t  : 4;                                      // Pad
  uint8_t  PID_GamePadMagnitude[4];                  // Usage 0x000F0070: Magnitude, Value = 32 to 100, Physical = (0 + 32) in 
  uint8_t  PID_GamePadDuration;                      // Usage 0x000F0050: Duration, Value = 32 to 8447, Physical = (0 + 32) in 10⁻² s units
  uint8_t  PID_GamePadStartDelay;                    // Usage 0x000F00A7: Start Delay, Value = 32 to 8447, Physical = (0 + 32) in 10⁻² s units
  uint8_t  PID_GamePadLoopCount;                     // Usage 0x000F007C: Loop Count, Value = 32 to 8447, Physical = (0 + 32) in 10¹⁶  units
} outputReport_t;
```

DJm00n, 2020-04-26:
Also I dumped HID descriptors from `xusb.sys` driver:
https://gist.github.com/DJm00n/07e1b7bb21643725e53b16f45e0e7022
It is interesting that driver supports separate LT and RT axis under the hood but this functionality is disabled! There are two HID Gamepad and RacingWheel descriptors for that purpose.
Vibration support is only avalible via `IOCTL_XUSB_SET_STATE` (0x8000E008). :(

DJm00n, 2020-04-27:
Heh. I found a way how to make LT and RT separate axis via HID!

There is xusb22 driver option for that:
```
Windows Registry Editor Version 5.00

[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\xusb22\Parameters]
"GamepadTriggerUsage"=dword:00320035

```

[image]
[image]

nefarius, 2020-04-27:
This is absolutely beautiful, nice find! 😍