nefarius/ViGEmBus #48: How to safely uninstall Fireshock by ViGEm ?

0lm, 2020-06-08:
Hi there. Can someone tell me how I completely uninstaller Fireshock?
I basically would like to revert my system back to the time before I installed Fireshock. And no I dont have a System backup.
It cant be found in the Device Manager. It seems to only popup when I plug in a Controller. Is there a tutorial how to a safely remove it?

nefarius, 2020-06-08:
1) Not a helpdesk, there's literally links to the right places scattered all around
2) Don't mess with Windows directories or you'll get bitten, use the proper tools like [DriverStore Explorer](https://github.com/lostindark/DriverStoreExplorer/releases)

closed 2020-06-08 by nefarius

0lm, 2020-06-09:
I found these tutorials and I did what it was stated in them. BUt still, there were leftovers and there were no information about these leftovers. Thats why I asked if it is safe to deleted the stated folders/files.

nefarius, 2020-06-09:
I see, DSE will be your friend there, it uses safe Windows routines to get the job done.