nefarius/ViGEmBus #51: Fake DS3 controller doesn't work with bluetooth

lbladetr, 2020-07-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior (example):
1. Start program '...'
2. Click on '....'
3. Plug in device '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System details (please complete the following information):**
 - OS: [e.g. Windows 10 1803]
 - Feeder software: [e.g. VDX, DS4Windows, ...]
 - Driver Version: [e.g. 1.14.3.0]

**Additional context**
Add any other context about the problem here.

nefarius, 2020-07-20:
... seriously?

closed 2020-07-20 by nefarius

last0tt0man, 2020-10-19:
> I'm sorry for previus version of issue first of all,it happened because of some network bug.I'm trying to solve a problem with DS3 for a long time,maybe more people encountering with this problem.I can use my DS3(good quality replica) with usb cable but when i go for bluetooth connection it pairs with dongle but dont react pressing the buttons or moving the analog.
> I don't get any errors but for the possibility i upload log file.
> [image]
> [image]
> [Shibari.Dom.Server-20200720.log](https://github.com/ViGEm/ViGEmBus/files/4949266/Shibari.Dom.Server-20200720.log)
> 
> Maybe it cause from my DS3 but i am trying to fix it so if anyone can explain why this is happening i will be happy,thanks in advance.I upload some screenshot for if i can't explain myself very well.
> 
> Windows 10 v.2004
> Shibari DOM Server v.1.5.171
> Fireshock v3.0
> BthPS3 v1.24

Hocam bu mal herif anca dalga geçmeyi bilir çözüm yok. Piyasada bundan başka bu işle uğraşan da yok.

last0tt0man, 2020-10-19:
> I'm sorry for previus version of issue first of all,it happened because of some network bug.I'm trying to solve a problem with DS3 for a long time,maybe more people encountering with this problem.I [...]

istersen bende tüm fake controller lar ile bluetooth çalışan eski versiyon var. Kendim paketledim ve setup haline getirdim. Onu paylaşabilirim.