nefarius/ViGEmBus #53: Win7 First Boot Gamepads in Joy.cpl DS4Windows Ryochan Issue

shinra358, 2020-09-02:
https://github.com/Ryochan7/DS4Windows/issues/1414

Any way something can be done on this end regarding this issue? Windows 7 Ultimate 64 bit here.

I do not have quickboot because it is not a feature of win7. System restore is not on, nor is hibernation mode. So nothing is being stored to memory when my cpu is shut down.

When first starting windows (after their permanent controller feature was introduced, and turning on ds4windows for the first time nomatter with autoboot or manually, virtual xbox controllers are seen in the joy.cpl when no controller is connected and when ds4windows is not on permanent mode. I have to stop and restart ds4windows (the whole program) for everything to function correctly. If I just press stop/start, the output slots are not cleared inside ds4windows.

So is there any way that something on the driver end can be adjusted to fix this issue for win7?

nefarius, 2020-09-02:
I'm not sure I fully understand the issue and how to reproduce, even after reading the linked issue.

So you're describing, that virtual controllers automatically come back/remain there even after shutdown/reboot?

The ViGEm bus driver has no concept of restoring previously connected devices by itself; once the last process using the client API has exited/died all emulated devices get reported as missing a.k.a. get unplugged and remain unplugged.

Are you sure it's not another program in auto-start-up also using the ViGEm API?

Cheers

shinra358, 2020-09-02:
"So you're describing, that virtual controllers automatically come
back/remain there even after shutdown/reboot?"

Yes, and i have to shut ds4windows down and reopen in order for that to be
rectified. A dev insists something in my system is saving the controllers
but ive already went through all the steps in proving otherwise. I dont
have anything using the vigem drivers that i know of. No old scp drivers
either. I really think it's the program (ds4w) but one dev insists that it
isnt. So i didnt know what else to do. Making an ahk script to close and
reopen ds4w seems silly.

mika-n, 2020-09-02:
I'm that dev who insists that DS4Windows app cannot keep a virtual gamepad device "alive" :-)  

When DS4Windows app is closed (not sent to background but closed completely) then all Vigem objects are destroyed and virtual xbox360 gamepad devices are removed (C# app, so even when an app would fails to explicitly delete objects then .NET garbage collector would take care of it).

I suspect the issue is somewhere in Win7 settings (notorious fast restart, except Win7 should not have it?) because, based on logfiles, those virtual xbox360 gamepad devices somehow "survive" over cold reboots and apper in JOY.CPL gamepad list even when DS4Windows app is not autostarted (can't see how DS4Windows app and Vigem driver could even theoretically do it on their own).

Or maybe there really is another app or another instance of DS4Windows app launched autoagically in the background? It could explain the behaviour (except DS4Windows app checks if the process is already running and refuses to open the same process twice).

Veeeery weird (and interesting) case.

nefarius, 2020-09-02:
Ah, I see, thanks for the clarification. Well, there is indeed a few ideas left on how to troubleshoot this.

We could after such a boot with survived controllers for example look with [Process Monitor](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon) if there are open handles to the driver, to identify a potential candidate that spawns the controller through the official API nonetheless.

Be my guests on [Discord](https://discord.vigem.org/) if you favour more "instant" interaction 😛

nefarius, 2020-09-02:
And Fast Startup and all that new-age nonsense was added in Windows 8, it is indeed not available in Windows 7.

shinra358, 2020-09-03:
Here is my startup logfile for process monitor

nefarius, 2020-09-03:
I actually meant Process Explorer but this is an even better idea, oops 😅 thanks for the submission, I'll take a look.

nefarius, 2020-09-04:
I found no hint 🙁

shinra358, 2020-09-04:
Doh! I sent new pics in the link above in the ds4windows case which gives a
visual using xinput checker etc if youd like to have a gander

nefarius, 2020-09-21:
Has there been some Eureka moment in this case out of interest?

shinra358, 2020-09-21:
Not on my end. The new updates still have it in them.

Squall-Leonhart, 2021-05-12:
This is clearly the result of having added a Vigem compatible utility configured as a system service which it should not be.

Windows 7 has no concept of a persistent kernel state, the only way virtual controllers would be present on the next boot is if an an application has connected them.

closed 2022-01-03 by nefarius

shinra358, 2022-01-03:
win7 issue only. doesnt happen in win10.

Squall-Leonhart, 2022-01-03:
doesn't happen on 7.

shinra358, 2022-01-03:
literally the whole purpose of me making this thread because of the issue
being on 7 -_-

Squall-Leonhart, 2022-01-03:
its something your side.

shinra358, 2022-01-03:
nothing on my side. literally spent months speaking to these guys about the
issue. switched to win10 bcuz of kingdom hearts and issue is not on win10.
and considering support for win7 has ceased for the project, it's a win7
issue....