nefarius/ViGEmBus #58: Vigem 1.16.116 rumble problem (Dualshock 4 v2)

LuismaSP89, 2020-11-06:
Some problems when the Vigem 1.16.116 are installed on the system. When a rumble event comes on some games (Using different emulators) the controller freezes and the character keeps walking/shooting/whatever.  After uninstalling the Vigem, the problem was solved.

This reminds me a similar problem on another software called ScpServer:

https://github.com/nefarius/ScpToolkit/issues/39

The thing is that I need Vigem in order to use DS4Windows + Cemu emulator. In this emulator, the problem happens in every game with rumble functions.

On another emulators such as RPCS3, the problem happens also in the games. But if I uninstall Vigem (Since RPCS3 haves dualshock passthrough and it doesn't need an external software) the problem is solved again.

There are any plans for releasing a new Vigem version? The latest one was on Dec 19. Almost a year ago.

My specs:

Windows 10 x64
GTX 970
i7 4790k
16 GB DDR3

Thank you for this awesome software.

nefarius, 2020-11-06:
There was a major bug regarding rumble in the client SDK so if these solutions still build upon and outdated client lib they might suffer from what you report. As for a new release, it's in the works but it addresses other issues, no rumble ones since, well, there are none known 😉

LuismaSP89, 2020-11-06:
@nefarius So, you mean this rumble problems are goint to stay for long, since the wip next release are not going to fix this? :S bad news.

Masamune3210, 2020-11-07:
No, they are saying that it's an issue with the app that you are using and that the app needs an update, not the backend....

Masamune3210, 2020-11-07:
In other words, update your shiz 😃

nefarius, 2020-11-07:
@LuismaSP89 thanks for the specs of your machine but the software versions and origins of the tools you use would help a great deal in this case 😉

LuismaSP89, 2020-11-07:
Sorry, but I don't get the point.

The software I use is the latest RPCS3 and CEMU versions. Both of this are working fine since If I uninstall Vigem, everything works fine.

The problem begins when Vigem is installed on the system. The rumble problems begin. If I uninstall Vigem, everything goes normally.

Like I said, I use Vigem because is needed for the DS4 windows "also using latest version" and the DS4 windows is needed for CEMU to enable motion controls.

In other words:
Cemu + vigem + ds4windows = rumble causing freezes on the controller.

Cemu without vigen and ds4windows = rumble working but without motion controls.

Rpcs3 without vigem or ds4 windows = all works fine, since rpcs3 have a dualshock4 passthrough.

Rpcs3 + vigem installed on the system= rumble problems like on cemu.

LuismaSP89, 2020-11-07:
Ds4 windows 2.1.5:  https://github.com/Ryochan7/DS4Windows

Rpcs3 0.0.13-11178:
https://rpcs3.net/compatibility?b

Cemu 1.21.5:
https://cemu.info/

This problem is happening for long ago, and solves when Vigem is uninstalled. I used vigem for many years. And the problem suddenly beginned on the latest version. This is the reason I discarded the problem on the emulators. (Also it's too coincidence that happens in different emulators, and solves by uninstalling vigem)

LuismaSP89, 2020-11-08:
Ok, after many tries, seems fixed by uninstalling Vigem, clearing cache, and installing again. Sorry.

closed 2020-11-08 by LuismaSP89

nefarius, 2020-11-09:
Interesting, clearing what cache though? Might help others...

LuismaSP89, 2020-11-09:
System cache (using ccleaner) after uninstalling Vigem. Also I deleted the pagesys file and enable it again. But I don't know if this one is necessary.