nefarius/ViGEmBus #59: PARSEC ViGEmBus issues

Wanaco21, 2020-11-10:
Hi! My friend install Parsec to play online but it doesn't allow me to use my gamepad (the game pad is okay, it works in my computer). I asked in a Parsec forum and they told me to reinstall Parsec and install ViGEmBus manually. 
My friend did all of that but it still doesnt work (VigemBus doesn't appears in the devices manager)

nefarius, 2020-11-11:
Hello,

on what OS version and driver version?

Wanaco21, 2020-11-11:
Hi! Windows 10 64 bits, i really dont know too much about drivers and all that stuff. The main issue is that i installed Parsec (that should also contain the VigemBus driver but it didnt installed it) and i tried installing that Driver manually from a setup that the Parsec support guy sent me and it didnt worked. 
Windows 10 64 bits 10.0.19041.1

nefarius, 2020-11-11:
I see, is something happening when the setup from here (the Release page) is used? Or does it just finish successfully and still nothing works in Parsec?

Cheers

Wanaco21, 2020-11-11:
It just finish successfully and still nothing works in Parsec :(

nefarius, 2020-11-11:
Ah I see, can you throw the following file my way from the machine that hosts: `c:\Windows\inf\setupapi.dev.log`

Thanks

Wanaco21, 2020-11-11:
Should i upload that file here?

DuckerMan, 2020-11-14:
dude just upload it on pastebin.com

nefarius, 2020-12-22:
New setup has been released, please retry, if issue still relevant.

closed 2020-12-22 by nefarius