nefarius/ViGEmBus #62: Xbox One controller emulation

FancyBanana, 2020-12-05:
I would like to request addition of Xbox One controller emulation.

Dualsense, the new controller released by Sony for PS5 has adaptive trigger features that physically interact with fingers. It's possible to create many different effects like trigger pull resistance or _vibration_. With this new functionality it is possible to replicate the trigger vibration of the Xbox One controller. The is already an API for Dualsense but there is no known way to emulate Xbox One controller.

Shayano, 2021-02-09:
Yes it's a great idea and it would allow to use the "copilot" function available on Windows 10 through the xbox accessories app.
It would simplify the life of solo games with multiple controllers.

Myoko, 2021-02-13:
good idea!

Masamune3210, 2021-02-13:
This might actually be better in something like DS4Windows

FancyBanana, 2021-09-07:
Any updates on this issue?

nefarius, 2021-09-08:
Nope.

nefarius, 2022-06-25:
This is a topic for the successor.

closed 2022-06-25 by nefarius