nefarius/ViGEmBus #63: Nefarius Virtual Gamepad Emulation Bus Driver Setup Wizard ended prematurely

Nike2406, 2020-12-20:
Hello!
I ran into a problem while installing **ViGEmBusSetup**. Installation fails with [**error** ](https://ibb.co/DpLm3Sg)Nefarius Virtual Gamepad Emulation Bus Driver Setup Wizard ended prematurely (check error-link with img) =(

**OS**: Windows 10 pro 19042.685

**ViGEmBus**: 1.17.333
**FireShock**: 3.0.0.0
**Shibari**: 1.4.106
**BthPS3**: 1.2.4

nefarius, 2020-12-20:
Hello! Please provide the file `c:\Windows\inf\setupapi.dev.log` from the troubled system, thanks.

Nike2406, 2020-12-21:
Here [setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/5724105/setupapi.dev.log) 
But I already installed the patch by checking the file system via dism (DISM /Online /Cleanup-Image /RestoreHealth).
Never the less, thank you for support)

nefarius, 2020-12-21:
> But I already installed the patch by checking the file system via dism (DISM /Online /Cleanup-Image /RestoreHealth).

Pardon me, what do you mean by that?

nefarius, 2020-12-21:
I've reviewed the file; it reports that the driver has been installed successfully...? Is that also what you experience now?

Nike2406, 2020-12-21:
> 
> 
> I've reviewed the file; it reports that the driver has been installed successfully...? Is that also what you experience now?

Yes, I already installed ViGEmBusDriver successfully.
Before this happened, i tried to reboot pc, checked another versions ... and so on. After that that, i tried to check windows files by utility Dism via Command line, and, it's a miracle, it's successfully installed)

nefarius, 2020-12-21:
That's a cool hint to know, thanks! May we close this issue then? 😀

Nike2406, 2020-12-21:
You're welcome =)

reztek, 2020-12-21:
I`m having similar problems. I wonder if it`s something related to this:
(Per https://support.microsoft.com/en-us/help/4577671/windows-10-update-kb4577671)
Symptom - 
"When installing a third-party driver, you might receive the error, “Windows can’t verify the publisher of this driver software”. You might also see the error, “No signature was present in the subject” when attempting to view the signature properties using Windows Explorer."
Workaround - 
"This issue occurs when one or more of the following is present in a driver package:
An improperly formatted catalog file is identified during validation by Windows. Starting with this release, Windows will require the validity of DER encoded PKCS#7 content in catalog files. Catalogs files must be signed per section 11.6 of describing DER-encoding for SET OF members in X.690.
A driver catalog file extension is not one of supported extensions.
If this happens you should contact the driver vendor or device manufacturer (OEM) and ask them for an updated driver to correct the issue."

[image]

nefarius, 2020-12-21:
@reztek nice find! Odd, the CAT signature is indeed complaining, on the 2nd one, the Microsoft one appears to be fine. Maybe dual-signing the CAT is not allowed or breaks the way I did it. I'll bake a test-release you could help me test 😁

reztek, 2020-12-21:
@nefarius Absolutely!

nefarius, 2020-12-21:
@reztek [here's a version](https://downloads.vigem.org/projects/ViGEmBus/testing/v1.17.333/) where I simply included the original CAT files like I got them back from Microsoft, single signature, nothing else changed. Please give it a go 😁

reztek, 2020-12-21:
Unfortunately, no dice. I got error "Windows Installer installed the product. Product Name: Nefarius Virtual Gamepad Emulation Bus Driver. Product Version: 1.17.333.0. Product Language: 1033. Manufacturer: Nefarius Software Solutions e.U.. Installation success or error status: 1603." in the System Log.
No "c:\Windows\inf\setupapi.dev.log" generated, though.

nefarius, 2020-12-21:
Oh dear. Could you try again first removing any older installation and closing down software that could potentially use it? Ideally after a reboot.

reztek, 2020-12-21:
Ok, found the problem (I guess):
[image]

nefarius, 2020-12-21:
Oh no... 😅 Compiled with dynamic runtime. Argh. Back onto the TODO 😬

reztek, 2020-12-21:
Ok, after installing the VC++ redist libraries, the ViGEm installer (release version) installed successfully. Thanks for your time (and software).

nefarius, 2020-12-21:
Nice job reporting that. I will fix these two issues next (invalid signature on CAT and the dependency on the runtime).

Cheers

nefarius, 2020-12-21:
I snuck in new setups in the existing release 😉

closed 2020-12-21 by nefarius

supplesummo, 2020-12-24:
I seem to have the same issue and am perhaps missing the trick to fix it described above. Can someone help?

nefarius, 2020-12-24:
Described what exactly? There have been a few suggestions. What did you try and what Windows version do you run under? When did you download the setup, before or after the fix has been uploaded?

JohnEdwa, 2020-12-26:
Hey, I encountered an error like this myself as well trying to update, and I managed to fix it.

The first time I ran the setup it told me to reboot and try again (I assume it had to uninstall the old one first), but after that reboot it started failing with this message every time.
I eventually went to the Device Manager and saw that it was creating a new "Nefarious Virtual Gamepad Emulation Bus" device with every try under "System Devices", with all but one of them showing the exclamation mark. Manually uninstalling all of them, rebooting, and running the setup again fixed it and allowed the update to happen. 

Here's that part from my log, in case it helps in the future somehow. 
[vigem_setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/5743314/vigem_setupapi.dev.log)

clin444, 2021-01-09:
Hi,
@JohnEdwa 's method, just uninstalling all the "Nefarious Virtual Gamepad Emulation Bus" drivers under "System Devices" worked for me as well.

Here's my log if it helps.

[setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/5790156/setupapi.dev.log)
(i took a look at the file and saw that parsec installs ViGEmBus, maybe that has something to do with it?)

Sch1nken, 2021-02-08:
Same thing for me, @JohnEdwa provides a working solution.

I also have Parsec installed, so that might be the problem. Though on a 2nd PC the same configuration works without issues.

elysi4n, 2021-04-28:
[image]
same issue

elysi4n, 2021-04-28:
[image]
 same issue

nefarius, 2021-04-28:
> [image]
> same issue

Can't help you without any logs, info which Windows build and architecture etc.

Proplayer584, 2021-04-29:
Hola, Tambien me aparece el mismo error al descargarlo que dice ended prematurely que puedo hacer?

Megustas44, 2021-05-28:
> [image]
> mismo problema

Hi, i have the same problem

OS: 1809 {17763.1879}
[setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/6561272/setupapi.dev.log)

PoggersPlayer, 2021-08-09:
> Here [setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/5724105/setupapi.dev.log)
> But I already installed the patch by checking the file system via dism (DISM /Online /Cleanup-Image /RestoreHealth).
> Never the less, thank you for support)

It won't fix the error!!

PoggersPlayer, 2021-08-09:
I WON'T STOP REFUSES TO WORK!

Masamune3210, 2021-08-09:
Are you completely nuts or do you think spamming comments is going to get anything other than the issue closed....