nefarius/ViGEmBus #65: stuck with driver and cannot uninstall

GlydetheSurfer, 2020-12-22:
[image]
 when trying to uninstall this happens and i cannot uninstall the driver.. im stuck with it

nefarius, 2020-12-22:
Exact Windows version and build number?

nefarius, 2020-12-22:
Do you have this file on your system `C:\Windows\System32\advapi32.dll` and if so, what version? See screenshot:

[image]

dejdyla, 2020-12-22:
Hi, having the same issue. My version of file mentioned above: "10.0.19041.610"

nefarius, 2020-12-22:
*Sigh* great, what broke now...

nefarius, 2020-12-22:
I'll provide a new (hopefully fixed) setup in a few. In the meantime [please use this Microsoft tool](http://support.microsoft.com/mats/Program_Install_and_Uninstall) to get rid of the now broken installation.

nefarius, 2020-12-26:
@dejdyla @GlydetheSurfer [could you please test these](https://downloads.vigem.org/projects/ViGEmBus/testing/) and report back. Bear in mind [you still need to rip out the broken installation first](http://support.microsoft.com/mats/Program_Install_and_Uninstall).

Cheers

nefarius, 2020-12-31:
I'd need some feedback on this one 😬

dejdyla, 2020-12-31:
worked for me. Thank you!

nefarius, 2020-12-31:
Sweet, thanks. Gonna swap out the official downloads soon 😊

dejdyla, 2020-12-31:
Oh sorry, the official one worked somehow worked for me. I actually haven't tried one you send here. Will try them ASAP.

nefarius, 2020-12-31:
I've updated the MSIs - yet again - on the release page. Hopefully for the last time now 😅 

Happy New Year! 🎉

GlydetheSurfer, 2021-01-01:
> @dejdyla @GlydetheSurfer [could you please test these](https://downloads.vigem.org/projects/ViGEmBus/testing/) and report back. Bear in mind [you still need to rip out the broken installation first](http://support.microsoft.com/mats/Program_Install_and_Uninstall).
> 
> Cheers

yah um.. im not installing it again thanks.. it started making my mouse lag

nefarius, 2021-01-01:
> > @dejdyla @GlydetheSurfer [could you please test these](https://downloads.vigem.org/projects/ViGEmBus/testing/) and report back. Bear in mind [you still need to rip out the broken installation first](http://support.microsoft.com/mats/Program_Install_and_Uninstall).
> > Cheers
> 
> yah um.. im not installing it again thanks.. it started making my mouse lag.. and I had to do a system restore

Having a hard time believing that... The driver just being present on the system literally does nothing 🤔

AdriandeLima, 2021-01-03:
I'll test it out soon and report back here

bsget, 2021-01-07:
I am having this issue with both the x64 version as well as x86. I also tried the two you posted to vigem.org with no success. The windows tool to clean out installations was not able to find any remaining registry files from the install. The failure page I got from the install was "Nefarius Virtual Gamepad Emulation Bus Driver Setup Wizard ended Prematurely" which is a page on the wizard itself. It says it made no changes to my system.

I'm happy to provide more information or test things if needed.

Information:
- advapi32.dll file version: 10.0.18362.752
- Windows version: 10.0.18363 Build 18363

bsget, 2021-01-08:
After installing visual studio 2019, the desktop c++ environment, and the wdf the installation was successful.

nefarius, 2021-01-08:
@bsget so the **installation** failed on you, not the **removal**? With the MSIs provided in the latest release?

bsget, 2021-01-08:
Yes, I had VS 2019 installed already, but it was an older version. I went to install with the MSI and it failed on me. I installed the newer VS 2019 (including the desktop dev with C++ workload) and also installed the WDF. Then restarted and re-ran the installer MSI and it worked.

There were no issues with the removal and I never had to use the removal tool you linked as the MSI installer failed gracefully.

nefarius, 2021-01-08:
Very weird, so the VCRedist package is missing and it still complains. Good to know, will address in next release. Wanna get rid of the devcon binary anyway.

GlydetheSurfer, 2021-01-10:
> Having a hard time believing that... The driver just being present on the system literally does nothing 🤔

well ...I didn't have any issues until after installing it. I'm guessing it was conflicting with another controller emulation driver I installed before

nefarius, 2021-01-10:
What do we do with this issue here, I consider it fixed?

GlydetheSurfer, 2021-01-10:
> What do we do with this issue here, I consider it fixed?

ok then

closed 2021-01-10 by GlydetheSurfer