nefarius/ViGEmBus #66: Rumble events lasts too much time in 1.17.333

LuismaSP89, 2020-12-26:
Using newer version of driver (1.17.333) "Tested with Cemu emulator" when a rumble event comes in, the rumble is extended too much time. For example, a rumble event that originally is only a hit or a shoot "0,25 seconds" the controller rumbles for 3-4 seconds. This is a bit annoying in games with too much rumble events.

Previously I used 1.6 version with some issues like controller disconnected problems when the rumble event come in. In this case, this is now solved, but this problem replaced the old one.

Using PS4 Dualshock V2.
Windows 10 x64 H2 Build.

nefarius, 2020-12-26:
I don't know Cemu very much, is it using the client SDK or are you using some other software to feed the virtual controller?

LuismaSP89, 2020-12-27:
Hello, I'm using the vigem driver + ds4 windows. But this problem started when I installed the latest version of vigem.

nefarius, 2020-12-27:
Which exact fork and version of DS4Windows? Info, info, info!

LuismaSP89, 2020-12-27:
Sorry. I'm using the latest version. You can download it from here:

https://github.com/Ryochan7/DS4Windows/releases

I don't know what more info can you need, but in case there's something more, please tell me.

And thanks for your help! : )

nefarius, 2020-12-27:
I see, I shall investigate! I tested the release with Dolphin Emulator and there it behaved superb. I'm confident we get this resolved!

LuismaSP89, 2020-12-27:
Hello again,
I figured you can test this not only in cemu emulator, it happens also with the proper ds4 windows itself.

Inside ds4 windows, go to profiles and edit the current profile "or create a new one if you don't have it" then, in the right side, go to "other" and select "test heavy" or "test light"  to test the rumble. When you stop the test, it keeps the rumble for 3-4 seconds more. So, is not a cemu only problem. Must be something in the latest vigem.

LuismaSP89, 2020-12-27:
[image]

Ryochan7, 2020-12-27:
@LuismaSP89 That test is **completely independent** of ViGEmBus. More info would be needed as to what version of DS4Windows and which controller you are using.

All I can gleam from the screenshot is that it looks like you are using a pure black (0, 0, 0) lightbar color. People posted recently about a Force Feedback bug that only occurred when using pure black as the lightbar color. That issue was fixed in the latest release of DS4Windows (2.1.23).

LuismaSP89, 2020-12-27:
I already posted the info. I'm using the latest version of ds4windows "2.1.23" and a ps4 dualshock V2.

In this case, I tested it and it doesn't matter about the lightbar color. The ds4 windows requires vigem driver, so, that's the reason to put the bug here. With older versions of vigem, this doesn't happen.

nefarius, 2020-12-28:
Thanks for chiming in and the explanation, after the holidays I'll test the release as well. Currently I only own a first generation DS4 though if that's important.

LuismaSP89, 2020-12-28:
In this case I think the generation of the DS4 makes no difference, the V2 has minor changes in the battery, sticks, and other optimizations in the bluetooth, but in my case, I use the controller wired, so, it doesn't matter.

LuismaSP89, 2020-12-28:
Ok... I deleted the profile of DS4 Windows, created it again, set the lightbar other than 0,0,0, and now is working. Was not working before when I changed the lightbar.

My apologies man, sorry for wasted time. Closing.

closed 2020-12-28 by LuismaSP89

nefarius, 2020-12-28:
Whew, am glad mate, thought I made it worse with the update while it should make it rock-solid 😅 Game on!