nefarius/ViGEmBus #67: v1.17.333: Windows 10 only 😭 ?

Darthagnon, 2020-12-31:
Wish Win7 installation was still permitted. Guessing that it's because of low-level black magic, and not as simple as signing a driver (or installing an unsigned one).

I'm holding off on updating drivers for the moment due to #65

nefarius, 2020-12-31:
I am using Microsoft dependencies not available on anything lower than 10 now, sorry mate. One guy alone can't stomach that project without some outside help and MS puts and end to 7 and 8.x soon so I'll go with the flow.

nefarius, 2020-12-31:
> I'm holding off on updating drivers for the moment due to #65

Well, without feedback and help from people suffering from this issue I can't fix it. Not able to replicate on any of my system so we gotta wait.

Darthagnon, 2020-12-31:
Perfectly understandable; wishing you a happy new year!

nefarius, 2020-12-31:
Likewise 😁 stay safe and healthy, cya in 2021!

closed 2021-02-12 by nefarius