nefarius/ViGEmBus #71: 1.17.333 keeps Ryzen Master from loading

flashmasta, 2021-01-08:
**Describe the bug**
When the service is running Ryzen Master will lock up when executed.

**To Reproduce**
Steps to reproduce the behavior (example):
1. Install 1.17.333, make sure the service is running
2. Run Ryzen Master, watch it lock up and then kill it
3. stop the ViGEmBus service
4. Open Ryzen Master and watch it work fine

**Expected behavior**
I uninstalled 1.17.333, reinstalled 1.16.116, and then it went back to working.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System details (please complete the following information):**
 - OS: Windows 10 20H2
 - Feeder software: DS4Windows but it crashes whether the app is open or not, the service is the only thing that needs to be running to reproduce
 - Driver Version: 1.17.333

**Additional context**
Ryzen Master is a very useful app from AMD for system monitoring and overclocking.

Masamune3210, 2021-01-08:
i feel like this is either a local issue with your config or a local incompat, ryzen master on latest works fine for me

flashmasta, 2021-01-08:
I wish that were true, but the only variable I have to work with is the new 1.17.333 service.  Disabling it fixes the issue.  This is on a 100% clean install of Windows, all drivers and software fully up to date, because I was racking my brain trying to figure out why Ryzen Master stopped working.

closed 2021-01-08 by flashmasta