nefarius/ViGEmBus #73: Assertion failed: Could not find 'git' via 'C:\Windows\System32\where.exe'

EdwardElric64, 2021-01-10:
Whenever I attempt to compile the driver powershell spits out the error "Assertion failed: Could not find 'git' via 'C:\Windows\System32\where.exe'" I installed everything asked of me in the pre requisite tab and restarded my PC to no avail. Why am I getting this error?

nefarius, 2021-01-10:
How do you compile it? The PowerShell scripts are meant to run on CI so I have no idea if they even work locally.

EdwardElric64, 2021-01-10:
In powershell I use the command that is on the page, ".\build.ps1 -configuration release"

nefarius, 2021-01-10:
Works a treat over here, mate. Make sure your build environment has all the dependencies. Or build within Visual Studio.

[image]

closed 2021-02-07 by nefarius