nefarius/ViGEmBus #74: Cannot uninstall, "This application can only be installed on Windows 10."

Lilikura, 2021-01-13:
I cannot currently uninstall the latest release of ViGEmBus x64-- I was forced for unrelated reasons to perform a rollback to a previous point and when I did so ViGEmBus started showing issues, notably crashing repeatedly, sometimes as much as 5-10x an hour. I assumed the update had been one of the things rolled back and attempted to reinstall it, only to find that I am now getting the error I can only install the application on Windows 10. (For note, I am on Windows 10.)

nefarius, 2021-01-13:
What do you mean by "crashing repeatedly", hard BSOD? [Does this work for you in regards of removal?](https://github.com/ViGEm/ViGEmBus/issues/65#issuecomment-751360749)

nefarius, 2021-01-13:
Found a lead on this issue.

nefarius, 2021-02-26:
Note: with development of BthPS3 I collected some new insights on more robust Windows 10 detection, will integrate for this project as well and test and afterwards close this issue.

closed 2021-03-30 by nefarius