nefarius/ViGEmBus #76: Controller Freezes on input

Cyberb28, 2021-02-07:
Describe the bug:

On older hardware computers your controller tends to freeze, I noticed because I was using a older computer and was experiencing constant freezing, I went onto my more powerful computer had no issues and once running a stress test my controller froze. 

**To Reproduce**
Steps to reproduce the behavior (example):
1.) Start the program
2.) Use a controller and plug it in
3.) run a stress test
4.) Controller freezes

**Expected behavior**
I am not quite sure what Im supposed to put here but maybe optimize the emulation software for older computers and that could possibly lessen the freezing if not stop it 

**System details (please complete the following information):**
 - OS: [Windows 10 Pro 20H2
 - Feeder software: Dualsense Controller CFI-ZCT1W
 - Driver Version: 1.17.333

**Additional context**
I noticed that this version only supports windows so I am curious over whether or not if I used a older version of ViGEmBus would help.

nefarius, 2021-02-07:
Sorry but I am not aware of a "Dualsense Controller CFI-ZCT1W" feeder software, what do you mean? Are you running SHibari by any chance?

Cyberb28, 2021-02-07:
My apologies, I looked at other threads an saw what I am supposed to add, I am using ViGEmBus

nefarius, 2021-02-07:
I get that but you're using ViGEmBus with **what**? The driver on its own does nothing, it requires feeder software to do anything useful, so once again; [are you maybe talking about this](https://github.com/ViGEm/Shibari)?

Cyberb28, 2021-02-07:
Oh yeah sorry I do have that, I followed the instructions from your website, I download all 3, ViGEmBus, SHibari and Fireshock I believe it was I have all 3, I am using SHibari