nefarius/ViGEmBus #79: [Question] Does your program support touchpad swipes now?

idgas01, 2021-02-24:
[image]

I have only been playing games on PlayStation Now for around 10 days now.  And I have only tried one program to emulate a PC keyboard and mouse.  It's called DS4Emulator, and it is easy to use and works great.  And actually, it requires that I install ViGEmBus to use it.  So it's already installed.

From the GIF screenshot above, you can see that I need to touchpad swipe.  I never needed to use a touchpad before, and I never owned a console controller before so I don't know much about this.  I"m thinking that DS4Emulator might let me touchpad click (but not swipe) and it probably only works in a select few games.

I searched on Google for "touchpad swipe Playstation emulate" and I found some one year old threads >>> https://github.com/ViGEm/ViGEmBus/issues/11 <<< that said that viGEmBus was working on it, but it hasn't been released yet.  That was one year ago.  Then I found a thread on your GitHub that said "Need to merge the SDK and .NET lib branches to master, then this issue is done."  That post was a few months ago.  So I'm thinking that ViGEmBus supports touchpad swipes now?  If so, how do I use it?  I haven't the faintest clue because even though ViGEmBus is a prerequisite for using DS4Emulator, I have only been using the DS4Emulator.

Your help would be greatly appreciated, and hopefully soon before my PlayStation Now subscription ends.  Thank you.

nefarius, 2021-02-24:
It does with the start of version v1.17, yes. That alone will do you no good though, as stated ViGEmBus is a framework, it will not magically start using new features, the people writing feeder software for do. There is new APIs both for C and dotnet live that allows setting the full capabilities of a DS4. Again, if you're not a developer, that information alone won't get you far.

I don't know the program you mentioned so I recommend contacting the author who'd need to update their code to utilize the new features. 

Cheers

idgas01, 2021-02-24:
Okay, thanks.  I'll close this issue now.

closed 2021-02-24 by idgas01

idgas01, 2021-03-02:
So um, as I mentioned before I am using DS4Emulator.  It's an AMAZING program.  I like it so much better than reWASD, despite it only having text config files instead of an interface menu.  The one thing that it is lacking is touchpad swipes.  That's it.  And not every game requires touchpad swipes, so that's good.

But anyway, I asked the developer of DS4Emulator if he could add touchpad swipes using ViGEmBus 1.17.  At first, his response was "Hi, I will look soon."  But then a day or two later, he conceded "I have tried adding this, but so far without success."  Would you be able to offer him any assistance?

He goes by the name r57zone.  His email is r57zone[at]gmail.com.  And his GitHub is https://github.com/r57zone/DualShock4-emulator.  Oh and finally, this is the GitHub thread https://github.com/r57zone/DualShock4-emulator/issues/14 where I asked if he could add the touchpad swipes.  If none of these options are doable for you, then do you have a permanent Discord invite link that I can share with him?  Maybe I can persuade him to join your Discord and ask for help there.

My PlayStation Now subscription ends soon and I probably won't renew it anytime soon, and the games that I do play in the next week will probably not require touchpad swipes.  But I'd still like to see his amazing program evolve to the next level.  And then maybe you can list his program in your list "Known users of ViGEm".

nefarius, 2021-03-03:
> I have tried adding this, but so far without success.

Can mean literally anything 🙂 you both are very welcome to join the Discord or forums for chatting about this topic. Or just provide more details in one of the mentioned issues. 

I don't engage in mail conversations about these topics since the content will remain private and of no use to the public which is the whole point of FOSS contributions.