nefarius/ViGEmBus #82: Nefarius Virtual Gamepad Emulation Bus Driver Setup Wizard ended prematurely

e-maximilian, 2021-02-26:
[image]
[setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/6049824/setupapi.dev.log)

nefarius, 2021-02-26:
[Please provide a detailed setup log](https://forums.vigem.org/topic/290/troubleshooting-a-failed-vigem-bus-setup) and attach it here. Thanks.

nefarius, 2021-02-26:
Exact window version and build number?

e-maximilian, 2021-02-26:
Edition	Windows 10 Pro Insider Preview
Version	2004
Installed on	‎28/‎01/‎2021
OS build	21301.1010
Experience	Windows Feature Experience Pack 221.501.0.3

Processor	ARMv8 (64-bit) Family 8 Model 0 Revision   0   1.00 GHz  (2 processors)
Installed RAM	1.99 GB
Device ID	7B9E6511-DC02-4656-8C35-7EBF482D2253
Product ID	00331-20090-00000-AA425
System type	64-bit operating system, ARM-based processor
Pen and touch	Pen support

I'm using Parallels Desktop 16.3.2 (50531)
MacBook Pro M1

e-maximilian, 2021-02-26:
[install.log](https://github.com/ViGEm/ViGEmBus/files/6050723/install.log)

nefarius, 2021-02-26:
Ah, I see, sorry mate, ARM machines are currently not supported! Only good old x64 and x86.

e-maximilian, 2021-02-26:
Any plan to support ARM?

nefarius, 2021-02-26:
> Any plan to support ARM?

I neither have experience nor hardware to test a build, so the chances are low. I've created #83 as a reminder but I can't make any promises.

nefarius, 2021-03-03:
Closed because unfixable for the time being.

closed 2021-03-03 by nefarius