nefarius/ViGEmBus #83: Attempt to create ARM64 builds

nefarius, 2021-02-26:
Apparently ARM64 installations are on the rise, the solution/project needs this platform integrated and tested.

nefarius, 2022-06-26:
[Here's the first **unsigned and untested** ARM64 build](https://buildbot.vigem.org/builds/ViGEmBus/master/1.18.378.0/bin/arm64/) for anybody who knows how to test-sign, manually install and put the machine in test mode.

nefarius, 2022-06-29:
We successfully installed and tested a build on a MacBook Pro 14 (M1 Pro) and everything seems to work fine! Next step is making a setup for it.

nefarius, 2022-08-08:
[An ARM64 **BETA** release is vailable](https://github.com/ViGEm/ViGEmBus/releases/tag/v1.19.418.0) whoever has the hardware, please test and provide feedback here, thanks.

CodeMonkeyInit, 2022-08-08:
Attempting to install on m1 mac air with windows 11 vm inside of parallels getting error: "The file 'nefconw.exe' is not marked for installation"

nefarius, 2022-08-08:
[Try this one please](https://downloads.vigem.org/projects/ViGEmBus/testing/ViGEmBus_1.19.418_x64_x86_arm64.exe).

CodeMonkeyInit, 2022-08-08:
> [Try this one please](https://downloads.vigem.org/projects/ViGEmBus/testing/ViGEmBus_1.19.418_x64_x86_arm64.exe).

double checked, getting same error with this one

nefarius, 2022-08-08:
> > [Try this one please](https://downloads.vigem.org/projects/ViGEmBus/testing/ViGEmBus_1.19.418_x64_x86_arm64.exe).
> 
> double checked, getting same error with this one
> 
> p.s managed to install by using /unpack command and manually installing driver with ARM64/ViGEmBus.inf

Well, *partial* progress is better than no progress 😉 We'll figure it out.

ynohtnalude, 2022-08-09:
not sure if it helps but I also get the 
"The file 'nefconw.exe' is not marked for installation"

Edition	Windows 11 Pro
Version	22H2
Installed on	‎2022-‎05-‎27
OS build	22621.160
Experience	Windows Feature Experience Pack 1000.22632.1000.0

Processor	Snapdragon(TM) 835   2.21 GHz
Installed RAM	4.00 GB (3.69 GB usable)
System type	64-bit operating system, ARM-based processor
Pen and touch	Pen and touch support with 10 touch points

It's a Lenovo Miix630

ImSaaam, 2022-08-09:
> > [Try this one please](https://downloads.vigem.org/projects/ViGEmBus/testing/ViGEmBus_1.19.418_x64_x86_arm64.exe).
> 
> double checked, getting same error with this one
> 
> p.s managed to install by using /unpack command and manually installing driver with ARM64/ViGEmBus.inf

I am getting the same error as well, can I ask you how you installed it?

nefarius, 2022-08-09:
Replaced the setup on the release page with a fixed one, please try again.

Delstrom2, 2022-08-10:
> [An ARM64 **BETA** release is vailable](https://github.com/ViGEm/ViGEmBus/releases/tag/v1.19.418.0) whoever has the hardware, please test and provide feedback here, thanks.

I don't know why this doesn't seem to be working on Parallels, but this works perfectly on my original Windows Arm hardware. You guys are lifesavers!

nefarius, 2022-08-10:
> I don't know why this doesn't seem to be working on Parallels, but this works perfectly on my original Windows Arm hardware. You guys are lifesavers!

Isn't parallels emulating a different CPU architecture?

Delstrom2, 2022-08-10:
> Isn't parallels emulating a different CPU architecture?

Not really, or at least it shouldn't be. The M1 is an ARM-based CPU, and Parallels can't translate CPU architecture and perform well, so it has to use a version of Windows designed to be installed for ARM64 hardware like mine (Galaxy Book2). It should be worth noting that I'm referring to full Windows ARM64, not Windows RT or Windows Mobile.

nefarius, 2022-08-10:
> Not really, or at least it shouldn't be. The M1 is an ARM-based CPU, and Parallels can't translate CPU architecture and perform well, so it has to use a version of Windows designed to be installed [...]

I see. I have no experience with Apple stuff or Parallels etc. so I'll take your folk's words for it 😇

CodeMonkeyInit, 2022-08-10:
> Replaced the setup on the release page with a fixed one, please try again.

Seems to be fixed now on Windows 11 inside of Parallels

ynohtnalude, 2022-08-10:
I tried this version posted on the 9th, it installed fine and x360 picked up the controller thanks!!

I don't know if the new one, posted on the 10th, to fix installation for parallels machines affects my machine

ynohtnalude, 2022-08-10:
I just open up xbox cloud gaming on my sd835 pc with windows 11 in MS edge,

My Logitech Rumble pad worked great playing forza5 on this wimpy machine!!

Thanks!!

ilyanengen, 2022-09-03:
> > Replaced the setup on the release page with a fixed one, please try again.
> 
> 
> 
> Seems to be fixed now on Windows 11 inside of Parallels 

Hi! I try to use with Windows 11 in Parallels, but after successfull installation I can't see Xbox 360 controller in Device Manager. I installed VigemBus 1.20.432.
What is the name of the device?
How can I check it installed correctly?

nefarius, 2022-09-03:
> Hi! I try to use with Windows 11 in Parallels, but after successfull installation I can't see Xbox 360 controller in Device Manager. I installed VigemBus 1.20.432. What is the name of the device? [...]

Well, what feeder software do you use? On its own there won't be any controller on the system.

ilyanengen, 2022-09-04:
> Well, what feeder software do you use? On its own there won't be any controller on the system.

What feeder software should I use? I will really appreciate if you could provide some examples.
PS: I installed XOutput, but it says it can't find VigemBus.

nefarius, 2022-09-04:
> What feeder software should I use? I will really appreciate if you could provide some examples.

That entirely depends on **your** use-case, I can't tell you that 😅 DS4Windows from Ryochan7 is the most prominent example.

jorbig, 2022-09-05:
> > What feeder software should I use? I will really appreciate if you could provide some examples.
> 
> That entirely depends on **your** use-case, I can't tell you that 😅 DS4Windows from Ryochan7 is the most prominent example.

But that one doesn't work with Windows on arm, right? Is there an alternative feeder software that does work?

driver1998, 2022-09-05:
Userspace feeder can just be x86/x64 and they will work fine. Also, DS4Windows is a .NET 6 app and shouldn't be too difficult to support ARM64.

ilyanengen, 2022-09-06:
> > Replaced the setup on the release page with a fixed one, please try again.
> 
> 
> 
> Seems to be fixed now on Windows 11 inside of Parallels 

Could you please share the name of feeder you used to make it work in Parallels? Thank you

nefarius, 2022-10-10:
Considering this done.

closed 2022-10-10 by nefarius