nefarius/ViGEmBus #84: the driver is unavailable

huseyinalperenn, 2021-03-02:
[image]

nefarius, 2021-03-02:
Good one, literally no helpful context provided whatsoever, easy fix 🙄

Ryochan7, 2021-03-02:
Have you tried turning it off and on again?

sergserg2, 2021-03-03:
I notice something similar. I'm using DS4 Windows. I think everything works fine when I start it for the first time. But if I restart (turn off/on connection in DS4 Windows) it locks up after a long while. Then the device cannot be turned off or removed. All that remains is to restart Windows.
This changed in last version of ViGEm, when fresh started after Windows boot is stable. Problem is later after reconnection. In earlier version this happens also on fresh started, but after long time (few hours).
 
[image]

nefarius, 2021-03-04:
Thanks @sergserg2 when that happens could you select the troubled device, then switch to View / Devices by connection and share that along with the error code in the General tab.

Cheers

closed 2021-03-17 by nefarius