nefarius/ViGEmBus #85: Driver not working on Windows Server 2019

benjaminburkhardt, 2021-03-16:
**Description**
I installed the latest version (ViGEmBus Setup 1.17.333) on a new Windows Server 2019 instance on AWS EC2.
Installation was successful and driver is visible in Device Manager. Unfortunately I cannot communication with the driver using the .NET API. 
My program is working perfect on my local Windows 10 machine with the same driver...

**To Reproduce**
1. Get new EC2 instance with Windows Server 2019
2. Install driver
3. try to .Connect() a new instance using .NET API.

**Screenshots**
[image]

**System details (please complete the following information):**
 - OS: Windows Server 2019
 - Feeder software: https://github.com/ViGEm/ViGEm.NET
 - Driver Version: 1.17.333

nefarius, 2021-03-16:
Server 2019 lacks the mandatory Xbox 360 drivers, have you installed them? Is the bus driver even present in Device Manager? Call stack?

benjaminburkhardt, 2021-03-16:
I just tried to install the Xbox 360 drivers from, here (https://www.techspot.com/drivers/driver/file/information/17648/) but this did not help. Is there another driver that I could try?

In the Device Manager I can see that the **Nefarius Virtual Gamepad Emulation Bus** is installed. And as long as my Visual Studio Application is not stopped and stuck at the .Connect() command the "Other Device" **Virtual Xbox 360 Controller** gets visible. Maybe that helps?
[image]

The Call Stack is not very helpful: 
[image]

Thanks a lot for your help!

nefarius, 2021-03-17:
Yeah, as I said, Xbox 360 Driver is missing so the virtual controller can't boot and therefore nothing will work as expected.

Please don't just randomly copy parts of drivers around like that, it's a brilliant way to test if you have a good backup 🙂

[Use the archived copy of the X360 driver instead](http://web.archive.org/web/20200425215425/http://download.microsoft.com/download/6/9/4/69446ACF-E625-4CCF-8F56-58B589934CD3/Xbox360_64Eng.exe).

benjaminburkhardt, 2021-03-17:
Great this was the missing piece! 😉 
Thank you!

closed 2021-03-17 by benjaminburkhardt