nefarius/ViGEmBus #91: Error Code 2503

akaXyle, 2021-04-10:
When I try to install the application I just get an error code.
I have tried installing 7 times now.
Windows 10 1909
Driver 457.51
[image]

nefarius, 2021-04-10:
That's an issue with Windows Installer that has popped up frequently this week for unknown reasons. The setup works as expected, yet unfortunately can't write to a temporary folder. Two methods have been reported by affected users to work:

- Run the MSI from an elevated command prompt (use search engine of your choice on how to do that)
- Validate/fix permissions of the TEMP folder, [like so](https://advyon.com/solved-2502-2503-error-when-installing-from-a-msi-file/) (just an example link, can't test myself because not affected)

Let me know if that worked. Good luck.

akaXyle, 2021-04-11:
I fixed the permissions of the TEMP folder and it worked! Thank you!

Rocky04, 2021-04-15:
The permissions for "C:\Program Files" and "C:\Windows\Temp" needs to be correct because the setup will first unpack the data into the Win temp folder and then copy them into the programs folder.

nefarius, 2021-04-15:
> The permissions for "C:\Program Files" (owner TrustedInstaller) and "C:\Windows\Temp" (owner SYSTEM) needs to be correct because the installer will first unpack the data into the Win temp folder and then copy them into the programs folder.

I wonder what broke it though?! The way the Windows Installer works (and the setup which builds upon it) hasn't changed in a long time, did a Windows Update alter permissions on accident? 

The setup can be invoked as non-Admin; once it requires elevated privileges the UAC will pop up. Somehow it already breaks at an earlier stage (I guess when it extracts the temporary copy of driver and devcon to prepare for installation).

Rocky04, 2021-04-15:
A bad system upgrade, very bad applications or bad user actions can cause this. I'm not aware of another thing which can mess with it.

nefarius, 2021-04-15:
> Wooting uses this software since a few days to emulate a controller to replace the Double Movement behavior in Fortnite so a deeper strafing angle can be archived in that game with any keyboard. So expect a lot more users due to the hype, because Epic seems to allow controller emulator nowadays.

Yay, more donors finally! 😆  (SCNR)

Kanuan, 2021-04-26:
# Step-by-step guide on how to fix Windows' `Temp` folder permissions

__The images and instructions used as examples may not match perfectly your experience, so do things accordingly to how your system presents itself on its language__

- Navigate to the `Windows` folder at the root of your local disk
- Find the `Temp` folder, right-click on it then select `Properties`  
[image]

- Go to the `Security` tab and then select `Advanced`
- Below the `Click Continue to attempt the operation with administrative permissions` message select `Continue`
[image] 

- Verify if the `Owner`, at the top, is `System`. If it's not, select the `Change` button at its right side, write `System` on the appearing window and then confirm  
[image]

- On the middle list, select the entry `Users (Your Username/Users)` and select `Edit`. At the right side of the appearing window, select `Show advanced permissions` and then leave only the following three permissions marked: `Traverse folder / execute file`, `Create files / write data` and `Create folders / append data`  
[image]

- Edit the `Administrators`, `SYSTEM` and `CREATOR OWNER` entries so they have `Full Control`  
[image]

- When all is done, confirm the changes in every window by selecting `OK`  

After Windows finishes applying the changes, try running Vigem's installer again and it should succeed now.

Rocky04, 2021-04-26:
The guide above should may include the note to disable inheritance for these folders if present and to check the option for replacing the permissions for the child elements for that so that the elements inside the folder get them too to avoid other problems.

**Overall, I think if people don't what to do it's may better to not mess around with these permissions at all because it can get worse.**

Kanuan, 2021-04-26:
This is the type of situations where I want to help but it's difficult to cover every possible scenario. I didn't want to just "go google it" for the reasons you mentioned: following a bad guide can lead the user to an even worse scenario.

This guide is based on top of how other users solved this on the discord server's support channel, where nobody yet appeared with _way too broken permissions_  and what is shown here worked as normal. I posted this here since I'm making a Vigem troubleshooting section and wanted to include this since it has become more frequent, so I just left a link from the article to here. 

All in all, I agree with you in that messing with permissions is dangerous when one doesn't know what they are doing. Let's see what are @nefarius thoughts on this. 

Depending on how this proceeds, I can:

- Just delete the whole guide
- Improve the guide here to try covering more edge-cases
- Leave things as how they are but leave a note saying that if there are too many things different from the guide it would better for the user to not follow it

Rocky04, 2021-04-26:
You misunderstood me and it should be obvious that I tried to improve your guide. There is no need to delete it.

I think it's good to have different views / opinions. There was a reason why I haven't shown in detail (with pictures) how to get to the permission view because I think that people who don't even understand the short description from me, should not mess with them anyway. But I also don't want to deal with explaining everything, like why there should be these settings and what they do. I think a good guide would do that kind of things because if a user follows them without knowing what they are actually do, the chance is high that they do something wrong. Also some people need a video because otherwise they don't understand that.

Also there must be a reason why some users have this problem and I think the most likelihood is that they used strange tools or they broke them by their own manually.

So again, your guide should include these things:
  - disable inheritance (so bad permissions from superior folders don't apply to them)
  - apply permissions within the container (to replace bad permission on child elements which are currently present)

Rocky04, 2021-04-26:
The first as you already updated it.

The inheritance from higher rank folder should be removed entirety, not converted. Maybe some people have applied some weird permissions on the entire partition. Or removed some for the entire Windows folder.

Squall-Leonhart, 2021-05-12:
doesn't the Uninstall/Install Fixit Wizard resolve this?

https://support.microsoft.com/en-us/topic/fix-problems-that-block-programs-from-being-installed-or-removed-cca7d1b6-65a9-3d98-426b-e9f927e1eb4d

Gluvii, 2021-06-05:
Thanks man.

closed 2021-08-14 by nefarius

Rocky04, 2021-08-15:
In the case someone is interesting why the file owner of the folders matter I think it comes down to User Interface Privilege Isolation.

The access-control list (ACL) of Win not only handles the file permissions, it also handles the Integrity Level (untrusted, low, medium, high & system). Every user or account has a default level and you can only see the current level in the permissions of a file or folder when the default level is manually overwritten / changed. So as a normal user the level of a normal started process is medium, but if the process is started as admin the level would be high. A process can't access files which have a higher level. A child process can have the same or a lower level (sandboxing) and so limited the potential access even the file permissions would allow it.

So if the installation is started within the system temp folder and the owner isn't system the level would be lower and so limited. Like installing a driver needs at least a high level so that the ViGEm bus driver can't be installed if the file owner is a user account and so only has a medium level.