nefarius/ViGEmBus #94: Renaming the setup binary

Rocky04, 2021-04-15:
Can you please rename the binary files to avoid confusion for people who don't get the wrong naming.

In the context both versions are in the x86 instruction set. They only differ by the bit length of the instructions, so the correct naming would be x86 32-Bit and x86 64-Bit.

Masamune3210, 2021-04-15:
x64, while not technecally correct (it should be x86-64 or AMD64) is commonly used around the internet to mean 64-bit. Even Microsoft does this with Windows.

Masamune3210, 2021-04-15:
Besides, explaining Intel's failure of a attempt to make IA64 happen would probably just confuse people more

Rocky04, 2021-04-15:
The main goal here is to differ between 32 and 64 bit. Now it's kind of weird due to a mixed naming convention. For completes x86 can still be used to differentiate other platforms. Like when an ARM build comes some day.

I recommend to avoid using AMD64 because then people who have Intel 64 are may confused because they don't know that they are compatible to each other.

nefarius, 2021-04-15:
I can't edit existing assets but I can adopt a new naming scheme in the future and offer adding a hint to the release notes which file name means what (which those who need the explanation probably won't read anyway but I digress) 😉 

[I even wrote a small article on the topic](https://vigem.org/research/How-to-check-architecture/) for a different project.

Rocky04, 2021-04-15:
I don't really get why to stick to the weird nomenclature from Microsoft. As mentioned if there will be a ARM build some day you should better provide the type and length anyway so why not start with it in the next version.

```
x86_32
x86_64
ARM32
ARM64
```

nefarius, 2021-04-15:
> As mentioned if there will be an ARM build some day you should better provide the instruction type and length anyway, so why not start with it in the next version.

As I said, I'll take it into consideration 🙂 And it will most probably change nothing because what experience showed me is that those who need to be told how to look up your CPU architecture won't read documentation anyway 😛 

Cheers

Rocky04, 2021-04-15:
So it's just me... I know what what it means and still don't like it.

So thanks for at least consider it.

closed 2022-06-25 by nefarius