nefarius/ViGEmBus #98: Mac support ?

vemito1, 2021-04-23:
Hey! 

I know this is currently only a windows 10 feature but I was wondering if your software will ever come to macOS! I’m hoping that it does!

nefarius, 2021-04-23:
Hello,

I personally am miles away from the knowledge required about development in an Apple environment, their driver ecosystem and have never owned any Apple hardware. So the signs are not the best 😅 but I wonder what benefit would it bring, Virtual Xbox Controllers on MacOS I assume?

vemito1, 2021-04-23:
Oh ok! It’s all good I know not many people dude software type of stuff on Mac thus it being limited to things that windows has that macOS doesn’t! 

Either way I really love this software and I can’t ware to see future inventions of yours

closed 2021-04-23 by vemito1

pixluser, 2022-11-25:
> 

It'll permit to a software like dualsensex to work on mac, and to bring support DualSense with Adaptative triggers… also with an universal method for developpers on Steam working on several Os I guess.

If a mac dev want to bring vigem on mac that'll be nice so :D