nefarius/ViGEmBus #102: Windows Server 2019 Reports Drivers for This Device Not Installed

jpflouret, 2021-05-05:
**Describe the bug**
When installed on Windows Server 2019 Datacenter and used with [RdpGameplad](https://github.com/microsoft/RdpGamepad) the Virtual Xbox 360 Controller device does not work properly. See [this bug report](https://github.com/microsoft/RdpGamepad/issues/17),

**To Reproduce**
- Install ViGEmBus
- Install RdpGamepad Receiver on server machine
- Install RdpGamepad Client on local machine
- Connect via Remote Desktop to server
- Observe Windows Device Manager reports driver for this device not installed (code 28)

**Expected behavior**
- Virtual Xbox 360 Controller works

**Screenshots**
See [this bug report](https://github.com/microsoft/RdpGamepad/issues/17).

**System details (please complete the following information):**
 - OS: Windows Server 2019
 - Feeder software: RdpGamepad
 - Driver Version: Unknown but assumed 1.17.333

**Additional context**
Add any other context about the problem here.

nefarius, 2021-05-05:
It's literally in the pinned issues (#85) good sir 🙂

jpflouret, 2021-05-05:
Bad developper. No cookie for me. Thanks.

closed 2021-05-05 by jpflouret