nefarius/ViGEmBus #105: Appears to Install but doesn't show in Device Manager or Function

marmski, 2021-05-27:
**Describe the bug**
I'm using this software as a part of the Wooting Double Movement for Fortnite. The software was working correctly until Wooting pushed an update of their software to my PC, then it immediately stopped working. I've uninstalled both Wooting and ViGEm and reinstalled in various ways (Wooting only didn't work, then ViGEm before Wooting and that didn't work, then ViGEm reinstall after Wooting install and didn't work). I can not see the Gamepad section in my Device Manager when I show hidden items but do remember it being there before the issue occurred. Also the Gamepad Tester website doesn't register a virtual gamepad being connected. I've also run all installations as administrator to attempt to rule out permissions issues.

I'm on Windows 10 - up to date with the latest packages.

I notice that I have a couple of PCI Devices that aren't functioning. Not sure what they are and unsure if that's related to what's happened here? (Image below)
Also notice that the Emulation Bus is appearing to be installed correctly in System Devices (Image below)

[image]
[image]

nefarius, 2021-05-27:
Your statements are contradicting each other; the screenshot clearly shows the bus device being present and healthy. For you PCI devices I am afraid you need to diagnose the state of your machine yourself 😉 Has nothing to do with the bus driver.

marmski, 2021-05-27:
Hey! Thanks for the prompt reply. I've had a read through quite a few threads on here trying to self-diagnose and noticed you're very active which is nice.

I agree - it appears to be functioning correctly but also isn't. It's very strange. Should it show a controller also connected under hidden items or should https://gamepad-tester.com/ show a gamepad connected? I'm using this software: https://github.com/WootingKb/wooting-double-movement/wiki/Troubleshooting and they appear to recognise that their update broke some people's program - it's just the solution they offer (which is to reinstall your software) doesn't solve it for me. Sorry if this isn't an issue for your page and should be instead directed at theirs - it was working perfectly and now I can't see any gamepads connected and the wooting software doesn't function. 

FYI: The other PCI issue wasn't related. I've sorted that now.

Appreciate any advice or support you can offer. :)

marmski, 2021-05-27:
Another update - I've done another set of reinstalls since updating my chipset drivers and now the gamepad appears to register in Gamepad Tester (Image below) but Fortnite doesn't pick up the inputs from the Wooting Software. I'll pick this up with Wooting on their page because it appears your side is now functioning correctly on my PC (unless you can see other issues I don't notice).

Appreciate the promptness of your replies. Wishing you the best.

[image]

nefarius, 2021-05-27:
Wireless Controller means you emulate a Sony DualShock 4. Is this intentional? I personally have no idea if Fortnite supports the DS4 and maybe you need to switch the Wooting Software to Xbox (X360) emulation?

Just guessing at this point; at its core ViGEmBus really isn't complex. If you have the bus driver installed and healthy, 99% of the work is already done. There are cases where problems can occur with X360 emulation in conjunction with the Microsoft `xusb22.sys` driver but those are rare and usually occur on systems that are.... broken beyond repair to put it mildly.

IIRC Wooting bundles the ViGEmBus setup with their software, so I guess just plainly attempting a re-install will fix a lot.

Other than that in case you experience that a virtual controller works in test utilities but not in games [you can check if you're accidentally hiding it](https://vigem.org/projects/HidGuardian/Taming-HidGuardian-Gen1/) but other than that there isn't really much to cover. If the driver is there and happy and the program in question uses the API correctly it works in 99.8% of the cases as advertised 😁 

Cheers

marmski, 2021-05-27:
Hey. So - I plugged in an Xbox (physical) controller to my PC and it went through installing the drivers (recognising the USB device) and suddenly everything started working as intended with the virtual controller again. Perhaps my Xbox controller drivers had become uninstalled? 

All functioning as expected now though. Thanks again for your help and prompt support.

nefarius, 2021-05-27:
Seems like the X360 driver was indeed missing in your case. Glad to hear it works! You may close the issue if you see it resolved.

closed 2021-05-27 by marmski