nefarius/ViGEmBus #106: Blocking for DInput devices

machomuchacho, 2021-05-30:
I'm emulating a DualShock4 controller with a Nacon Compact PS4 Controller and figured out while using Steam in Big Picture mode (for testing) that navigation felt super fast, I think it's due to the fact that the PC still seems to detect the Nacon controller because it moves 2 slots at a time every time I press a d-pad button or move the left stick, like if I was using 2 controllers at the same time. In UCR, [Interception](https://github.com/oblitum/Interception "Interception GitHub page") has an option to block the input from the physical device, is it possible to do the same with ViGEm ?

nefarius, 2021-05-30:
Sounds like a use-case for [HidHide](https://github.com/ViGEm/HidHide).

machomuchacho, 2021-05-30:
Perfect!

closed 2021-05-30 by machomuchacho