nefarius/ViGEmBus #114: Switch Pro controller Emulation Support

dude98, 2021-07-16:
It'd be awesome to have an emulated switch pro controller, as it has motion controls that work with Steam. It'd also allow the program BetterJoyForCemu to emulate a switch pro controller using joy cons, allowing the joy con motion controls to function on steam instead of emulating a mouse.

Ryochan7, 2021-07-16:
And reasons like this are why Valve has blocks in place for Steam Input when it detects other mappers running. Besides, the most common use case would be to map the Gyro to a virtual mouse anyway.

dude98, 2021-07-16:
while that'd most likely be the case, there's still a usefulness for it, like having switch icons appear in games that auto change the icons depending on the device used, or if a game doesn't allow keyboard plus gamepad input at the same time it'd still allow you to use motion controls without losing gamepad inputs.

MGM175, 2021-09-03:
Greetings. I also got the Nintendo Switch Pro gamepad, and I have a problem that emulated controller tend to stick to axis (usually right). I hope that implementation would be done. :D

[image]

nefarius, 2021-09-03:
> Greetings. I also got the Nintendo Switch Pro gamepad, and I have a problem that emulated controller tend to stick to axis (usually right). I hope that implementation would be done. :D
> 
> [image]

Your post has literally nothing to so with this request, please refrain from hijacking unrelated topics, thanks.

MGM175, 2021-09-06:
Ok, thanks for info.