nefarius/ViGEmBus #125: Setup Wizard ended prematurely : setupapi file marks error 0xe000022

BlueLight-yt, 2021-08-16:
I have an error, when I try to install Vigem, because before I had it on another pc and if it worked, this one gives me an error
[image]
[image]
[image]

Masamune3210, 2021-08-17:
Why do people insist on not reading logs that explain the issue, ignore the issue templates, and then expect help....

nefarius, 2021-08-17:
Nice, sure, add screenshots of text so it's extra hard to read or copy. Sometimes I really worry about the world...

Read 👏 And 👏 Follow 👏 The 👏 Templates 👏 And 👏 Get 👏 Out 👏 Of 👏 The 👏 Issue 👏 Tracker 👏

closed 2021-08-17 by nefarius