nefarius/ViGEmBus #128: ViGEmBus does not install on Windows 11 Build 22000

sjkorvin, 2021-09-05:
**Describe the bug**
I have tried using the 64-bit installer for version 1.17.333. The installer makes no progress. After a reboot, the same thing happens with version 1.16.116. I cannot find the vigembus.sys driver installed anywhere on my system.

**To Reproduce**
1. Open ViGEmBus installer .msi or .exe
2. Click to install
3. Approve UAC prompt

**Expected behavior**
I expect for the installer to complete and for ViGEmBus to successfully install.

**Screenshots**
I am currently at my parents' house. I can add screenshots later if necessary.

**System details (please complete the following information):**
 - OS: Windows 11 22000.176
 - Feeder software: Tried installing through latest Ryochan7 DS4Windows. Then tried with the executables hosted on this project's GitHub page 
 - Driver Version: 1.17.333 64-bit and 1.16.116

**Additional context**
I do not think that I have installed ViGEmBus through any other means. DS4Windows sends a notification to Windows saying that the ViGEmBus driver is not installed. I keep BetterJoy around, but I do not think that I have made use of it on this install of Windows.

nefarius, 2021-09-06:
Where are the logs? Why didn't you use the correct issue template?

[image]

sjkorvin, 2021-09-06:
I'm very sorry. I was very tired yesterday. I ended up sleeping all day today. I will try to be more careful in the future. Perhaps, I saw it, thought it was a link to the forums, and wasn't able to register what the proper thing to do was.

Please find attached the logs.

[install.log](https://github.com/ViGEm/ViGEmBus/files/7117839/install.log)
[setupapi.dev.log](https://github.com/ViGEm/ViGEmBus/files/7117840/setupapi.dev.log)

nefarius, 2021-09-07:
There we go, thanks. Hm, this is odd. From the setup log it just seems to end. No error, nothing. Cane you make screenshots or a screencast of how it looks when you click through it? And why do you need to reboot? I have no force reboot logic in the setup so wherever that is coming from, definitely not normal.

Cheers

sjkorvin, 2021-09-07:
The reason why I would reboot was to try the install again. I would start the installer and click to install. Then, no progress would be made in the installation. After closing the installer through Task Manager because I couldn't close it by clicking Cancel and then trying to start the install again, I had received a message saying that an installation was in progress and that the current installation would have to complete before starting a new one.

I just tried the install again and it worked! I tried to take a screencast of it, but I mistakenly stopped the screencast instead of ending it. It took some seconds for the UAC prompt to show up. Maybe, I had assumed that the UAC prompt had shown up and that I had clicked to accept it when it had not shown up. Either way, I had waited far longer than several seconds on previous installation attempts.

Would you like any logs?

nefarius, 2021-09-08:
So.... It worked now?

sjkorvin, 2021-09-08:
Yes, it worked now.

nefarius, 2021-09-08:
Welp, computers suck 😆 Congratulations! We can close this?

sjkorvin, 2021-09-08:
Yes! Thank you for your time and assistance!

closed 2021-09-08 by sjkorvin