nefarius/ViGEmBus #134: ViGEmBus doesn't install on Windows 11 Build 22621 (22H2)

sn-o-w, 2022-06-22:
**Describe the bug**
The installer just throws the error "Installer stopped prematurely."

**To Reproduce**
Steps to reproduce the behavior (example):
1. Open ViGEmBus installer
2. Click to install

**Expected behavior**
I expect the installer to successfully complete the setup.

**Screenshots**
Even better, a video:

https://user-images.githubusercontent.com/2406819/175100750-b87aa747-42f4-476f-9d6f-90961bdd370e.mp4

**System details (please complete the following information):**
 - OS: Windows 11 22H2 (22621.160)
 - Feeder software: Not exactly sure what is this, but I try to use it along with XOutput.
 - Driver Version: 1.17.333 64-bit

**Additional context**
I would like to mention that I don't even get the UAC prompt during the installation, so I'm not sure if this is important or not.
22H2 is the next Win 11 upgrade that is slowly released to Windows users, so perhaps your software needs some adjustments for 22H2? 🤔

sn-o-w, 2022-06-22:
Wow... I found out why the installation failed. It looks like .NET Framework 3.5 was missing.

I swear to God I had .NET Framework 3.5 installed previously, I think something got fucked up after upgrading to 22H2. 🤨

Perhaps you can add something in the installer that will warn the user if .NET Framework 3.5 is not installed?! 🤔

Should I close now the issue?!

nefarius, 2022-06-22:
I see what I can do 👍

sn-o-w, 2022-06-22:
@nefarius Read my previous message. 😂

nefarius, 2022-06-22:
> Perhaps you can add something in the installer that will warn the user if .NET Framework 3.5 is not installed?! 🤔

And you read mine 😛 my reply was in regards to this 😉

sn-o-w, 2022-06-22:
Oh, my bad then. I thought you missed my message regarding .NET Framework 3.5.

Squall-Leonhart, 2022-06-27:
Microsoft partially broke .net 3.5 in 22000 that required users uninstalling and reinstalling it, the same bug might have occured in 22621.

nefarius, 2022-06-27:
This topic should be a thing of the past with the new setup starting with version 1.18.x, closing.

closed 2022-06-27 by nefarius