nefarius/ViGEmBus #140: Weird Interactions with Focusrite Scarlett Drivers/Devices

Dmac9244, 2022-07-19:
**Describe the bug**
With both the ViGEmBus drivers and Focusrite drivers for their Scarlett 2i2 3rd gen interface installed, Windows experiences random audio cutout issues, where audio cuts out for around 2-5 seconds or longer. If I am in a Discord call, my microphone audio will be delayed by several seconds, so others in the call hear me several seconds after I say anything. It could usually be fixed by re-choosing the same microphone in Discord audio settings, but under specific circumstances (running Blue stacks emulating an Android game) Discord would *refuse* to be fixed until I quit the emulator. In one particularly egregious case, my audio *output* became delayed, so that I heard everything several seconds after it actually happened, while my *input* was on-time.

**To Reproduce**
Requires Focusrite Scarlett interface and drivers to reproduce
- Install Focusrite drivers
- Install ViGEmBus
- Make sure Windows is using Scarlett interface as audio output
- Play music or any kind of audio (tends to happen more often with greater CPU stress)
- Wait

**System details (please complete the following information):**
 - OS: Windows 10, Windows 11, the problem seems independent of sub-version because it has happened across two different versions of Windows
 - Feeder software: ??
 - Driver Version: Unknown, it was the version packaged with BetterJoy
 - Focusrite Scarlett driver version 4.102.4
 - NVidia RTX 3080
 - AMD Ryzen 3800x

**Additional context**
This seems to be a very specific issue that can't conveniently be replicated, but the problem was bugging me so much that I did a fresh install of Windows (and upgraded to Windows 11 in the meantime) to get rid of it. I wanted to be able to use BetterJoy again, so I installed the drivers again, and when I rebooted to initialize them my computer started experiencing the audio glitch within 10 minutes of the reboot. That's... not really a coincidence. Given, I'm only guessing that the problem is caused by some interaction with the Focusrite drivers, but it seems to me like a fair guess given that most people don't experience this problem, most people wouldn't have Scarlett interfaces, and on a practically fresh Windows 11 install, the most niche thing I'm running at the moment are those Focusrite drivers. If absolutely required I could try to uninstall the Focusrite drivers and see if that also fixes the issue, but I kind of *need* my interface to be able to use my computer normally. If it's any help, the problem seemed to be able to be fixed by switching Windows and Discord from the interface to another device.

nefarius, 2022-07-19:
You find the driver version by opening Device Manager, open System devices node and look for Virtual Gamepad Emulation and on the general tab you see the version, share it here. Have you tried running https://www.resplendence.com/latencymon

Cheers

TreetopMike, 2022-08-17:
I recently purchased a new windows 11 computer for myself, and one for my son (who uses discord, and has your identical problem).  I have found that i too have this issue when playing audio in another app, or using premiere pro (audio stalls and glitches as you describe).   We are both using the same focusrite driver (4,102.4).  Ive been using my focusrite interface for years on my old machine with no issues.  I have requested support from focusrite on the issue, but have so far come up dry.  I will be watching this post to see if anything develops.  If you find a solution, please post.  Thanks!

TreetopMike, 2022-08-17:
> You find the driver version by opening Device Manager, open System devices node and look for Virtual Gamepad Emulation and on the general tab you see the version, share it here. Have you tried running https://www.resplendence.com/latencymon
> 
> Cheers

I have run that latency monitor btw, and it suggests a driver or hardware conflict (wdf01000.sys spike) when the glitches happen.  Again, the glitches only happen in audio apps when the focusrite driver is installed.  If i uninstall, there are no issues.

nefarius, 2022-10-10:
IDK how to investigate this further.

closed 2022-10-10 by nefarius